enjoy-digital/litex #848: VideoTerminal uses 50k+ LUTS on DECA board

hansfbaier, 2021-03-13:
The on chip ROMs and RAMs are not synthesized into M9K memory blocks for some weird reason,
but into LUTs. I spent most of the whole day fiddling with the settings and the annotations,
but quartus stubbornly refused to infer block ram,
even when I manually annotated the array definitions in the verilog file with
ramstyle/romstyle annotations.
I checked that the annotations actually were applied, but it still would not infer block RAM/ROM nevertheless.

enjoy-digital, 2021-03-13:
Hi @hansfbaier,

I indeed also noticed that while adding support for it on the DECA but haven't been able to look at it yet, will try to do it soon.

hansfbaier, 2021-03-15:
Maybe it is because of this: [https://stackoverflow.com/questions/45052196/inferring-a-single-port-ram-in-verilog](https://stackoverflow.com/questions/45052196/inferring-a-single-port-ram-in-verilog)
[image]
[image]

hansfbaier, 2021-03-15:
I tried to pull the assign into the always-block as an experiment, but that made no difference :(

hansfbaier, 2021-03-15:
On Cyclone V (arrow SoCKit) results look good:
```
+-------------------------------------------------------------------------------+
; Fitter Summary                                                                ;
+---------------------------------+---------------------------------------------+
; Fitter Status                   ; Successful - Mon Mar 15 09:06:03 2021       ;
; Quartus Prime Version           ; 20.1.0 Build 711 06/05/2020 SJ Lite Edition ;
; Revision Name                   ; arrow_sockit                                ;
; Top-level Entity Name           ; arrow_sockit                                ;
; Family                          ; Cyclone V                                   ;
; Device                          ; 5CSXFC6D6F31C8                              ;
; Timing Models                   ; Final                                       ;
; Logic utilization (in ALMs)     ; 2,100 / 41,910 ( 5 % )                      ;
; Total registers                 ; 2552                                        ;
; Total pins                      ; 33 / 499 ( 7 % )                            ;
; Total virtual pins              ; 0                                           ;
; Total block memory bits         ; 356,576 / 5,662,720 ( 6 % )                 ;
; Total RAM Blocks                ; 63 / 553 ( 11 % )                           ;
; Total DSP Blocks                ; 4 / 112 ( 4 % )                             ;
; Total HSSI RX PCSs              ; 0 / 9 ( 0 % )                               ;
; Total HSSI PMA RX Deserializers ; 0 / 9 ( 0 % )                               ;
; Total HSSI TX PCSs              ; 0 / 9 ( 0 % )                               ;
; Total HSSI PMA TX Serializers   ; 0 / 9 ( 0 % )                               ;
; Total PLLs                      ; 1 / 15 ( 7 % )                              ;
; Total DLLs                      ; 0 / 4 ( 0 % )                               ;
+---------------------------------+---------------------------------------------+
```

hansfbaier, 2021-03-15:
No idea, why Max10 is such a stubborn troublemaker.

hansfbaier, 2021-03-18:
I just stumled over this: [https://retroramblings.net/?p=1509#more-1509](https://retroramblings.net/?p=1509#more-1509)
And it is the solution to the problem.
Will prepare a pull request.

enjoy-digital, 2021-03-18:
@hansfbaier: Thanks, interesting finding!

enjoy-digital, 2021-03-25:
We can now close this since fixed with https://github.com/litex-hub/litex-boards/pull/188, thanks @hansfbaier.

closed 2021-03-25 by enjoy-digital