enjoy-digital/litex #856: demo hangs on boot if it contains more than 16 bytes of .data

hansfbaier, 2021-03-19:
In the demo application, when defining a global array like this:
```
#define CRASHME 1
uint8_t some_random_data[] = {
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f,
	0xd0, 0x0f
#if CRASHME
	,0xd0
#endif
};
```
Then the application will hang on boot if CRASHME == 1 and successfully run, if CRASHME == 0.
In other words, if there are more than 16 bytes of global data allocated, the application will fail to boot (freeze at liftoff).
Reproduction: DECA board with `integrated_main_ram_size = 0x4000`

michalsieron, 2021-08-17:
From my tests it seems like it isn't array bigger than 16 bytes, but anything with size not divisible by 4.

michalsieron, 2021-08-18:
Adding `. = ALIGN(4)` before [demo/linker.ld#L34](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/demo/linker.ld#L34) fixed it for me. However I don't know why it works that way.

michalsieron, 2021-08-18:
Ok, I think I found definitive answer for that. `data_loop` ([crt0.S#L65](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/crt0.S#L65)), which copies `.data` section, works on words and advances pointers by 4. That's why the size needs to be multiple of 4 and aligning `_edata` to 4 fixed it.

PR #1007 (2021-08-23): Fix alignments in demo/linker.ld
Without this change, when `.data` section size wasn't multiple of word size, `data_loop` in crt0 was jumping over `_edata` and continued looping. As it works on words and right now 64 bit CPUs are biggest ones supported - alignment is now 8 bytes.

Also removed `- 4` from stack address, as it needs to be aligned to 16 bytes on RISC-V.

Should fix #856 and at least explain why #876 wasn't working.
closed 2021-08-23 by enjoy-digital