enjoy-digital/litex #869: Documentation/tutorial for using litex with litex-boards (and VexRiscv-SMP core) ?

Ruinland, 2021-04-03:
Hi,
sorry for bothering people here.

I'm wondering is there any guide for new comers to learn how to add a board to `litex-board` which exposes the interface for its user to configure the CPU features ?

Though being played around Linux-on-LiteX/VexRiscv for quite a long time,
still I didn't get my hands down to LiteX deep enough.

As far I as I know, the `litex-boards/litex/target/<board_name>.py` could be used to generate RTL (verilog) files, LiteX BIOS and other necessary firmware files.
Then a build directory will be created with a build script inside.

However, I'm not quite sure about how the CPU-specific arguments are passed down from board to SoC.
Specifically, I'm trying to generate the RTL with following configuration - -
cpu-type : VexRiscv SMP
cpu-count : 2
with FPU and RV-C Extensoin
on Lattice Versa ECP5 board.

With the educated guess learnt from Linux-on-LiteX/VexRiscv, I tried : 
`litex-boards/litex/target/lattice_versa_ecp5.py --cpu-type vexriscv_smp --cpu-count 2 --with-rvc --with-fpu`

Then the board file complains that it cannot understand the arguments :
`lattice_versa_ecp5.py: error: unrecognized arguments: --cpu-count 2 --with-fpu --with-rvc`

I looked around the `lattice_versa_ecp5.py` and didn't find out where args being passed down to `vexriscv_smp/core.py` .
"Not a big deal, I can modify core.py anyway", naively I thought.

So I change the default value of cpu_count in `class VexRiscvSMP(CPU)` from 1 to 3, and then I test it : 
`litex-boards/litex/target/lattice_versa_ecp5.py --cpu-type vexriscv_smp`

```
[info] Running (fork) vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=3 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=128 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=False --fpu=False --cpu-per-fpu=4 --rvc=False --netlist-name=VexRiscvLitexSmpCluster_Cc3_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Ood --netlist-directory=/home/ruinland/litex_2021_user/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog
[info] [Runtime] SpinalHDL v1.4.4    git head : b42ad071474c2042bf13d9ff30147b0e378b322d
[info] [Runtime] JVM max memory : 1700.0MiB
[info] [Runtime] Current date : 2021.04.04 01:24:40
[info] [Progress] at 0.000 : Elaborate components
[info] [Progress] at 3.417 : Checks and transforms
[info] [Progress] at 5.155 : Generate Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] memReadAsync can only be write first into Verilog
[info] [Warning] 855 signals were pruned. You can call printPruned on the backend report to get more informations.
[info] [Done] at 6.360
[success] Total time: 11 s, completed Apr 4, 2021 1:24:47 AM
```

It seems that the verilog files are generated successfully.

So I take the shot again, modifying `with_fpu` and `with_rvc` in `class VexRiscvSMP(CPU)` from `False` to `True`.
However, it fails to generate the verilog file : 
```
VexRiscv cluster : VexRiscvLitexSmpCluster_Cc2_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Ood_Fpu4_Rvc
Generating cluster netlist
[info] Loading settings for project vexriscv-build from plugins.sbt ...
[info] Loading project definition from /home/ruinland/litex_2021_user/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/project
[info] Loading settings for project root from build.sbt ...
[info] Loading settings for project spinalhdl-build from plugin.sbt ...
[info] Loading project definition from /home/ruinland/litex_2021_user/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/SpinalHDL/project
[info] Loading settings for project all from build.sbt ...
[info] Set current project to VexRiscv (in build file:/home/ruinland/litex_2021_user/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/ext/VexRiscv/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] Running (fork) vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=2 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=128 --aes-instruction=False --out-of-order-decoder=True --wishbone-memory=False --fpu=True --cpu-per-fpu=4 --rvc=True --netlist-name=VexRiscvLitexSmpCluster_Cc2_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Ood_Fpu4_Rvc --netlist-directory=/home/ruinland/litex_2021_user/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog
[info] [Runtime] SpinalHDL v1.4.4    git head : b42ad071474c2042bf13d9ff30147b0e378b322d
[info] [Runtime] JVM max memory : 1700.0MiB
[info] [Runtime] Current date : 2021.04.04 01:33:56
[info] [Progress] at 0.000 : Elaborate components
[info] **********************************************************************************************
[info] [Warning] Elaboration failed (0 error).
[info]           Spinal will restart with scala trace to help you to find the problem.
[info] **********************************************************************************************
[info] [Progress] at 0.289 : Elaborate components
[error] Exception in thread "main" java.lang.AssertionError: assertion failed
[error] 	at scala.Predef$.assert(Predef.scala:156)
[error] 	at spinal.core.package$.assert(core.scala:420)
[error] 	at vexriscv.ip.DataCacheConfig.<init>(DataCache.scala:41)
[error] 	at vexriscv.demo.smp.VexRiscvSmpClusterGen$.vexRiscvConfig(VexRiscvSmpCluster.scala:233)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$parameter$1.apply(VexRiscvSmpLitexCluster.scala:146)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$parameter$1.apply(VexRiscvSmpLitexCluster.scala:145)
[error] 	at scala.collection.generic.GenTraversableFactory.tabulate(GenTraversableFactory.scala:148)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.parameter(VexRiscvSmpLitexCluster.scala:145)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anon$1.<init>(VexRiscvSmpLitexCluster.scala:184)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.dutGen(VexRiscvSmpLitexCluster.scala:182)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$29.apply(VexRiscvSmpLitexCluster.scala:192)
[error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$29.apply(VexRiscvSmpLitexCluster.scala:192)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$2.apply$mcV$sp(Phase.scala:2169)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$2.apply(Phase.scala:2164)
[error] 	at spinal.core.internals.PhaseCreateComponent$$anonfun$impl$2.apply(Phase.scala:2164)
[error] 	at spinal.core.fiber.Engine$$anonfun$create$1.apply$mcV$sp(AsyncCtrl.scala:144)
[error] 	at spinal.core.fiber.AsyncThread$$anonfun$1.apply$mcV$sp(AsyncThread.scala:58)
[error] 	at spinal.core.fiber.EngineContext$$anonfun$newJvmThread$1.apply$mcV$sp(AsyncCtrl.scala:39)
[error] 	at spinal.sim.JvmThread.run(SimManager.scala:51)
[error] Nonzero exit code returned from runner: 1
[error] (Compile / runMain) Nonzero exit code returned from runner: 1
[error] Total time: 5 s, completed Apr 4, 2021 1:33:57 AM
```

However, I have built the bitstream with such configuration in `Linux-on-Litex/Vexriscv` successfully.
Thus I'm sure that there must be some big mistakes in my attempt way to do this.

I know this is a complicated combination of various projects hosted by different organization piecing together.
So I might being asking a question not belonging here, I apologize in advance for that (and the lengthy letter as well).

Sincerely,
Ruinland

Dolu1990, 2021-04-03:
@Ruinland 

> VexRiscvLitexSmpCluster_Cc2_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw128_Ood_Fpu4_Rvc

This is a unvalid configuration, basicaly, the FPU require the data cache to support 64 bits load/store. Here is is set to be 32 bits (Dw32 mean DataWidth=32)

Here is how the config is made legal when the FPU is enabled : 
https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv_smp/core.py#L99

enjoy-digital, 2021-04-06:
@Ruinland:  With the targets from LiteX Boards, it's only possible for now to select the `cpu-type` and` cpu-variant`. VexRiscv-SMP allows more specific configurations, but this is only exposed on Linux-on-LiteX-VexRiscv project. We could list the options available for a specific CPU, but not sure it would be possible with argparse (ex: ./arty.py --cpu-type=vexriscv-smp --help could list the variants for VexRiscv-SMP but also the specific configuration parameters).

Ruinland, 2021-04-08:
@Dolu1990 : 
Thanks for the tips! :+1:  Sorry for my dumb questions.

@enjoy-digital : 
I see. Thanks for the clarification :-) I'll close this issue.
Yet I might reopen it some other time in the future.

Recently, I've bought some more cheap boards from QMTech, ALINX and Terasic with a bit more LUTs than ECP4CE15 (namely, EP4CE22/EP4CE22/......) and I'm planning to support them with LiteX.
I guess that I'm gonna need help then 🤓

closed 2021-04-08 by Ruinland