enjoy-digital/litex #876: litex/vexriscv boot problem

andreapas-ap, 2021-04-09:
Hello, I am working on a custom board based on Artix xc7a100tcsg324-1, Litex/vexriscv, code flash memory and DDR RAM. System is properly working, but in some rare builds I am getting the following behavior (from debug UART):

``` Booting from flash...
Copying 0x90500000 to 0x40000000 (59241 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--
```

Nothing wrong about that, but unluckily user software is not starting at all, as something wrong is happening just after BIOS liftoff.
Additional note: I have a static initialized variable in my C code, something like ```static uint8_t my_var = 1;```. If I change it to uninitialized, so ```static uint8_t my_var;```, then boot problem disappears. No problem at all also if I change variable from initialized uint8_t to initialized uint32_t, while initialized uint16_t bevavior is same as for initialized uint8_t. 

Thanks in advance for the help

Andrea

enjoy-digital, 2021-04-14:
Hello @andreapas-ap,

thanks for the feedback. The issue could be similar to #856 but could also be related to an issue with the DRAM calibration. Do you  notice changes in the calibration logs between builds that are working and build that aren't? Could you eventually share the calibration logs for both cases?

andreapas-ap, 2021-04-15:
Hi, here both console logs:

- first from firmware defining uint8_t initialized static variable, user software not running after boot:

```        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr  9 2021 11:11:13
 BIOS CRC passed (743fc585)

 Migen git sha1: --------
 LiteX git sha1: --------

--=============== SoC ==================--
CPU:        VexRiscv @ 100MHz
BUS:        WISHBONE 32-bit @ 4GiB
CSR:        8-bit data
ROM:        1024KiB
SRAM:       8KiB
L2:     8KiB
SDRAM:      262144KiB 16-bit @ 800Mbps/pin

--========== Initialization ============--
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |01111111111111100000000000000000| delays: 07+-07
m0, b09: |00000000000000000111111111111110| delays: 24+-07
m0, b10: |00000000000000000000000000000000| delays: -
m0, b11: |00000000000000000000000000000000| delays: -
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b08 delays: 08+-07
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |01111111111111110000000000000000| delays: 08+-07
m1, b09: |00000000000000000111111111111111| delays: 24+-07
m1, b10: |00000000000000000000000000000000| delays: -
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b08 delays: 08+-07
SDRAM now under hardware control
Memtest at 0x40000000...
[########################################]
[... 81 lines cut ...]
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b08 delays: 08+-07
SDRAM now under hardware control
Memtest at 0x40000000...
[########################################]
[########################################]
Memtest OK
Memspeed at 0x40000000...
Writes: 212 Mbps
Reads:  188 Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
Booting from flash...
Copying 0x90500000 to 0x40000000 (59244 bytes)...
[########################################]
Executing booted program at 0x40000000

--============= Liftoff! ===============--
I:main:437: FPGA size: 100
I:main:443: firmware built Apr 15 2021 15:07:45 from git 88d44b6
```

enjoy-digital, 2021-08-23:
Should be fixed with https://github.com/enjoy-digital/litex/pull/1007.

closed 2021-08-23 by enjoy-digital