enjoy-digital/litex #878: Digilent zedboard + litex + VexRiscV

fredyamalves, 2021-04-12:
I am trying to get VexRiscv working on the Digilent zedboard. I found out that there is a "platform" file for it on "litex-boards" but not a "target file" and I understand that it is because the Zedboard does not have a UART connected to the FPGA itself, instead it is hardwired to the ARM.

I found out that you can actually make the UART accessible for the FPGA by making it an output port of the ARM using something called "EMIO". I exported both "UART 0" and "UART 1", added VexRiscV to the block design and connected its tx and rx to one of the UARTs at a time so I could test if any of them would work.

When I upload the bitstream to the board, VexRiscv seems to be working since it is blinking the LEDs sequentially, but when I try to run the provided "rotating donut" demo using:

litex_bare_metal_demo --build-path=build/digilent_zedboard/
lxterm /dev/ttyACM0 --kernel=demo.bin

I do not get any response. Here is an image showing how I connected the RTL (PL) to the processing system (PS):

Could anyone help me with this issue? Thank you.

[image]

tcal-x, 2021-04-12:
@fredyamalves , does the UART work simply to connect to the LiteX BIOS?   If you run
```
lxterm /dev/ttyACM0
```
and hit return a few times, do you see the `litex>` prompt (possibly after seeing the LiteX banner and initilization info) ?

fredyamalves, 2021-04-12:
@tcal-x I do not see anything, not even the Banner, would that mean the UART is not compatible with litex or could it be a problem with the design on the board? I have used this UART before with a baud rate of 115200 8ne1

tcal-x, 2021-04-12:
Ok, we've determined that the issue isn't with loading the donut demo, it's the basic UART connectivity.   Unfortunately I don't have any experience with Zynq's PS/PL stuff with LiteX.   You can try asking on the #litex IRC channel -- it is probably a quicker way to figure  out what is currently supported.

fredyamalves, 2021-04-12:
Thank you @tcal-x , I will join the channel and ask there.

closed 2021-04-12 by fredyamalves