enjoy-digital/litex #888: BIOS TFTP netboot not happy with everything in boot.json

nickoe, 2021-04-21:
# Background
I was trying to simulate a design where I used a `boot.json` that I would boot from a SD card, so I decided to try the TFTP netboot option, but it appears that it does not work well if there are more than on entry in the `boot.json`.

# Issue description
Take for example:

```
{
        "litex_setup.py": "0x41000000",
        "demo.bin":       "0x40000000"
}
```

`litex_setup.py` just being a simple file I am sure you have somewhere as a standing for some random data.

It appears to get stuck on that with just:
```
...
Booting from network...                                                                                                                                                                                                                      Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Copying litex_setup.py to 0x41000000...
```

If I just have the `demo.bin` application in the `boot.json` it will happily boot it.

if I swap the order, such that it looks like:
```
{
        "demo.bin":       "0x40000000",
        "litex_setup.py": "0x41000000"
}
```

It will load the `demo.bin` and get stuck on the `litex_setup.py` again:
```
...
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Copying demo.bin to 0x40000000... (8780 bytes)
Copying litex_setup.py to 0x41000000...
```

I tried tftp-hpa and ptftpd. The latter with this command:
```
sudo -E /home/yourock/.local/bin/ptftpd tap0 .
```

Running the sim with:
```
lxsim --with-ethernet --with-etherbone
```

And generating the app with:
```
litex_bare_metal_demo --build-path build/sim
```

Litex commit: b55af2156ba3380a7bed9eafeb33c64304179440

enjoy-digital, 2021-04-26:
Thanks @nickoe for the repro. I've been  able to reproduce it with your simulation and will look at it. The issue is related to use of the hybrid mode of the LiteEth MAC (both connected to the CPU (`--with-ethernet`) and to a hardware IP/UDP stack (`--with-etherbone`). I think I hit the same issue on hardware a few weeks ago, being able to reproduce it in simulation should simplify things :)

enjoy-digital, 2021-04-27:
This has been fixed with https://github.com/enjoy-digital/liteeth/commit/6febb1aaabfb8ad83bc1c4de2d366410dae63ed5.

closed 2021-04-27 by enjoy-digital

nickoe, 2021-04-27:
@enjoy-digital Thank you very much, I can confirm that it appears to resolve the issue :)