enjoy-digital/litex #890: Pull request #884 seems to break nexys video sdram

andrewb1999, 2021-04-25:
On the current head of master I get the following issue with the digilent_nexys_video.py target in litex-boards:
```
--=============== SoC ==================--                                                                                                                                                                   
CPU:            VexRiscv @ 100MHz                                                                                                                                                                            
BUS:            WISHBONE 32-bit @ 4GiB                                                                                                                                                                       
CSR:            32-bit data                                                                                                                                                                                  
ROM:            64KiB                                                                                                                                                                                        
SRAM:           8KiB                                                                                                                                                                                         
L2:             8KiB                                                                                                                                                                                         
SDRAM:          524288KiB 16-bit @ 800MT/s (CL-6 CWL-5)                                                                                                                                                      
                                                                                                                                                                                                             
--========== Initialization ============--                                                                                                                                                                   
Initializing SDRAM @0x40000000...                                                                                                                                                                            
Switching SDRAM to software control.                                                                                                                                                                         
Write latency calibration:                                                                                                                                                                                   
m0:0 m1:0                                                                                                                                                                                                    
Read leveling:                                                                                                                                                                                               
  m0, b0: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b1: |00000000000000000011111111111111| delays: 09+-22                                                                                                                                                  
  m0, b2: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b3: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b4: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b5: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b6: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m0, b7: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  best: m0, b01 delays: 09+-22                                                                                                                                                                               
  m1, b0: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m1, b1: |00000000000000000011111111111111| delays: 09+-22                                                                                                                                                  
  m1, b2: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m1, b3: |00000000000000000000000000000000| delays: -                                                                                                                                                       
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 09+-22
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  0/256
  addr errors: 8191/8192
  data errors: 524287/524288
Memtest KO
Memory initialization failed
```

But when I switch back to commit 0ed7852779fd0d8185e784b103116ca21cecb571 everything works fine, which leads me to believe #884 is causing this issue. If it helps with reproduction, I am using Vivado 2020.2.

jedrzejboczar, 2021-04-26:
There was an error in the wrapping logic and causing it to select delay of 9 instead of ~25, https://github.com/enjoy-digital/litex/pull/895 should fix the problem.

enjoy-digital, 2021-04-26:
Thanks @andrewb1999 for reporting, The wrap around has been removed with https://github.com/enjoy-digital/litex/commit/48ec20e2efdb5ed6a48a36c7c49f12ac6e40c8a9 and will fix the issue. We'll work on this together with @jedrzejboczar to see if it's really required and if so, be sure that all aspects are taken into account.

closed 2021-04-26 by enjoy-digital