enjoy-digital/litex #893: mor1kx bios region `rom' overflowed by 420 bytes

stffrdhrn, 2021-04-25:
With master from 2021-04-26 I am seeing this overflow.

I have bisected it to commit 80bd4ac4ecd756db755b1f758f0b59052402340c. 

```
bios: Add boot command to be able to boot directly from system memory.
```

This is when building the openrisc using litex-boards with the following options
## Build 
```
./litex_boards/targets/digilent_arty.py --cpu-type mor1kx --cpu-variant linux+smp+fpu --csr-json /home/shorne/work/openrisc/or1k-utils/litex/tftpd/litex-mor1kx-csr.json --with-ethernet --csr-data-width 32 --with-sdcard --no-compile-gateware --build
```

## Output
```
< shorne@lianli ~/work/litex/litex > make -C /home/shorne/work/litex/litex-boards/build/digilent_arty/software/bios -f /home/shorne/work/litex/litex/litex/soc/software/bios/Makefile V=1
make: Entering directory '/home/shorne/work/litex/litex-boards/build/digilent_arty/software/bios'
or1k-elf-ld -nostdlib -nodefaultlibs -L/home/shorne/work/litex/litex-boards/build/digilent_arty/software/include -T /home/shorne/work/litex/litex/litex/soc/software/bios/linker.ld -N -o bios.elf \
        ../libbase/crt0.o \
        isr.o boot-helper.o boot.o helpers.o cmd_bios.o cmd_mem.o cmd_boot.o cmd_i2c.o cmd_spiflash.o cmd_litedram.o cmd_liteeth.o cmd_litesdcard.o cmd_litesata.o main.o complete.o readline.o \
        -L../libcompiler_rt \
        -L../libbase \
        -L../liblitedram \
        -L../libliteeth \
        -L../liblitespi \
        -L../libfatfs \
        -L../liblitesdcard \
        -L../liblitesata \
        -llitedram -lliteeth -llitespi -lfatfs -llitesdcard -llitesata -lbase-nofloat -lcompiler_rt
or1k-elf-ld: bios.elf section `.data' will not fit in region `rom'
or1k-elf-ld: region `rom' overflowed by 420 bytes
make: *** [/home/shorne/work/litex/litex/litex/soc/software/bios/Makefile:67: bios.elf] Error 1
make: Leaving directory '/home/shorne/work/litex/litex-boards/build/digilent_arty/software/bios'
```

dayjaby, 2021-04-26:
@stffrdhrn is the bios size larger than 64KB? If yes, can you try to set --integrated-rom-size to a larger value like 128KB?

stffrdhrn, 2021-04-26:
That does help with the bios build, but the place n route fails later:

```
python3 -m litex.soc.software.memusage bios.elf /home/shorne/work/litex/litex-boards/build/digilent_arty/software/bios/../include/generated/regions.ld or1k-elf

ROM usage: 64.41KiB     (50.32%)
RAM usage: 2.28KiB      (28.52%)

make: Leaving directory '/home/shorne/work/litex/litex-boards/build/digilent_arty/software/bios'
```

```
ERROR: [Place 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 128 of such cell types but only 100 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
```

I will try to see if I can get the bios under the limit by changing how the function introduced in 80bd4ac works.

enjoy-digital, 2021-04-26:
Thanks @stffrdhrn for reporting, it seems the ROM was already very close to the default maximum size (64KB) with this configuration previously:
```
ROM usage: 63.43KiB 	(99.11%)
RAM usage: 2.28KiB 	(28.52%)
```
And adding this extra feature made it pass the 64KB. That would be nice in the future to have metrics to avoid the ROM usage to increase too much in the time (and maybe define some BIOS features as optional), but to fix your case, as described by @dayjaby this can be done by passing `--integrated-rom-size=0x20000 `. LiteX is now able to auto-resize the ROM at the end of the build, so since 64KB does not seems to be enough for your case, we could increase it to 128KB by default.

enjoy-digital, 2021-04-26:
@stffrdhrn: With Vivado 2018.2 and LiteX upstream, I've been able to build the design with: `./digilent_arty.py --cpu-type mor1kx --cpu-variant linux+smp+fpu --with-ethernet --csr-data-width 32 --with-sdcard --build --integrated-rom-size=0x20000` That's strange we have different results. Are you using the latests repositories for the cores? (https://github.com/enjoy-digital/litesdcard/commit/52e47a2bcffcadaa82c468efbdeb8732e81b33af#diff-a2b8ec40302d716f25af97ec87fe2e16079f8ab340cb1248028ff94545ef18d3 probably impacts resource usage).

Otherwise, for https://github.com/enjoy-digital/litex/commit/80bd4ac4ecd756db755b1f758f0b59052402340c, maybe it's not optimal, but the fact it broke the build is more an artifact and related to the fact the ROM was already almost full. (Just want to say that there are probably optimizations to do in the BIOS code and the added boot features is maybe not the one where we can gain the more).

stffrdhrn, 2021-04-26:
I updated the repo's yesterday.

Perhaps the issue is my old version of vivado, I am using 2017.3.  I will try to update it.

I was concerned about updating the ROM from 64k to 128k as I thought it would double the block FPGA ram usage.  But is that not the case?  Will it just use the minimum required memory?

enjoy-digital, 2021-04-27:
@stffrdhrn: In fact the specified `integrated_rom_size` is now a maximum and an auto-resize is done at the end of the build to reduce it to the minimum. This auto-resize is only done for a true ROM (`integrated_rom_mode="r")`, for a ROM that can be updated (integrated_rom_mode="rw"), as you were using for example to update the BIOS over JTAGBone, the auto-resize will not be done.

When running:  `./digilent_arty.py --with-ethernet --cpu-type=mor1kx --with-sdcard --with-ethernet --integrated-rom-size=0x20000`

You'll see:
```
INFO:SoC:Initializing ROM rom with contents (Size: 0x10100).
INFO:SoC:Auto-Resizing ROM rom from 0x20000 to 0x10100.
```

So the ROM is reduced to 0x10100 here.

stffrdhrn, 2021-04-27:
Thanks that makese a lot of sense now.  I confirm my rom setting is/was set to `integrated_rom_mode="r"`.  In the mean time I have tried to upgrade to vivado 2018.3 but running into issues.

```
****** Vivado v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source digilent_arty.tcl
# create_project -force -name digilent_arty -part xc7a35ticsg324-1L
ERROR: [Common 17-539] Failed to load tclapp initializer can't find package ::tclapp::support::appinit 1.1 from '/home/shorne/.Xilinx/Vivado/2018.3/XilinxTclStore'
ERROR: [Common 17-69] Command failed: ERROR: [Common 17-539] Failed to load tclapp initializer can't find package ::tclapp::support::appinit 1.1 from '/home/shorne/.Xilinx/Vivado/2018.3/XilinxTclStore'
```

Once I get the build working I will create a patch updating the ROM limit to 128k and we can close this.

stffrdhrn, 2021-04-28:
Updating to Vivado `2019.2` worked.  I couldn't get `2020.x` to work as I don't have enough disk space and `2018.3` failed for the above mentioned issue.

PR #900 (2021-04-28): integration/soc_core: Increase default ROM size to 128K
With recent BIOS changes the default rom size on the mor1kx built just
passes over the 64k boundary and the build fails.  Since the 128K
default is a soft limit and the ROM gets resized to the actual ROM
requirements this should be relatively safe.

Note, that if a RW rom is configured the full 128kb will be allocated.

Fixes: #893
commit 2f96cf02 (2021-04-28): integration/soc_core: Increase default ROM size to 128K

With recent BIOS changes the default rom size on the mor1kx built just
passes over the 64k boundary and the build fails.  Since the 128K
default is a soft limit and the ROM gets resized to the actual ROM
requirements this should be relatively safe.

Note, that if a RW rom is configured the full 128kb will be allocated.

Fixes: #893
closed 2021-04-28 by mithro