enjoy-digital/litex #894: litescope_cli can't work with RemoteClient class

jimmymagemtek, 2021-04-26:
Hi Sir,
I tried to invoke litescope_cli.py, to grab  dump.csv from analyzer modules on  fpga via litex_server.py
However, I got the following error messages:

=======================================================
[root@localhost software]# ./litescope_cli.py
Traceback (most recent call last):
File "./litescope_cli.py", line 138, in <module>
main()
File "./litescope_cli.py", line 120, in main
analyzer = LiteScopeAnalyzerDriver(bus.regs, basename, debug=True)
AttributeError: 'RemoteClient' object has no attribute 'regs'
[root@localhost software]# =======================================================

Do I miss any thing to make it work ?
Do I need to modify "RemoteClient" class manually?

Please help.
Many thanks.

Jimmy

enjoy-digital, 2021-04-26:
Hi @jimmymagemtek,

when the `csr.csv` is present in the local directory, RemoteClient will not be able to create the regs attribute and will then allow writing/reading to/from addresses specified by the user. Can you check you have the `csr.csv` of your design in the directory where you run the `litescope_cli` command?

Regards,

Florent

enjoy-digital, 2021-04-26:
If the `csr.csv` is present, can you add some print traces here: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_client.py#L23-L31?

jimmymagemtek, 2021-04-26:
Hi Florent,
Thanks for the prompt response.

I found that "litescope_cli.py" & "litex_client.py" use "RemoteClient" class in the same way, so  I move "litescope_cli.py" to the folder where "litex_client.py" resides & do some minor modification. Then the "regs" issue is gone, although I encounter another issue (some attributes, like trigger_enable not defined in "LiteScopeAnalyzerDriver" class, https://github.com/enjoy-digital/litescope/issues/32)

Do you still need the trace ?

Jimmy

enjoy-digital, 2021-04-26:
Thanks @jimmymagemtek for the feedback, we'll discuss the remaining issue in https://github.com/enjoy-digital/litescope/issues/32.

closed 2021-04-26 by enjoy-digital

jimmymagemtek, 2021-04-28:
I found that the "RemoteClass"  issue is about missing csr.csv, as you mentioned earlier. Sorry for my earlier wrong comment about folder change.

dayjaby, 2021-06-03:
Hey Jimmy,

can you try this PR whether it solves this problem for you: https://github.com/enjoy-digital/litex/pull/927

Cheers
David