enjoy-digital/litex #898: UARTWishboneBridge or litex_server may be broken

DurandA, 2021-04-28:
I updated an old project to use the latest LiteX, so I updated all LiteX dependancies but the UART wishbone bridge does not work anymore.

I previously used the old interface:
```python
self.add_cpu(UARTWishboneBridge(self.platform.request("serial"),
    sys_clk_freq, baudrate=115200))
self.add_wb_master(self.cpu.wishbone)
```
and I updated  the interface:
```python
bridge = UARTWishboneBridge(self.platform.request("serial"), sys_clk_freq, baudrate=115200)
self.submodules.bridge = bridge
self.add_wb_master(bridge.wishbone)
```

However, using `RemoteClient` with `litex_server` always produce `socket.timeout: timed out` errors.

I reduced my bench to a minimum in order to identify the issue:
```python
from migen import *

from litex.soc.cores.uart import UARTWishboneBridge
from litex.soc.integration.builder import Builder

from litex_boards.targets.ecp5_evn import BaseSoC

class LiteScopeSoC(BaseSoC):

    def __init__(self):
        sys_clk_freq = int(60e6) # check

        BaseSoC.__init__(self, sys_clk_freq, x5_clk_freq=int(50e6), toolchain="trellis",
            cpu_type=None,
            csr_data_width=32,
            with_uart=False,
            with_timer=False
        )

        # bridge
        bridge = UARTWishboneBridge(self.platform.request("serial"), sys_clk_freq, baudrate=115200)
        self.submodules.bridge = bridge
        self.add_wb_master(bridge.wishbone)        

    def do_exit(self, vns):
        pass

soc = LiteScopeSoC()
builder = Builder(soc, csr_csv="test/csr.csv")
vns = builder.build()

# Create csr and analyzer files
soc.finalize()
soc.do_exit(vns)
```

However,  it still fails. The UART is working (I can interact with the LiteX BIOS using a SoC with the same target). I tried with a minimal example as in [fpga_101/test_leds](https://github.com/litex-hub/fpga_101/blob/master/lab003/test/test_leds.py). Calling `wb.regs.leds_out.read()` generates a `socket.timeout: timed out`.

Was the combination `UARTWishboneBridge` + `litex_server` tested recently? Did I make a stupid mistake?

Thanks.

enjoy-digital, 2021-04-28:
@DurandA: This is regularly tested and I'm currently using it while doing some work on LiteDRAM. In LiteDRAM bench this is added with `self.add_uartbone(...)`: https://github.com/enjoy-digital/litedram/blob/master/bench/xcu1525.py#L106.

I see you have a `#check` near sys_clk_freq, the issue could be related to this: If correct sys_clk_freq is not passed to UARTWishboneBridge, it will not work.

DurandA, 2021-04-28:
I made a really stupid mistake. I didn't realize LiteX also changed the build directory structure so I was flashing an outdated gateware that was still present. :expressionless:

@enjoy-digital I am very sorry for wasting your time. Thank you for your help.

closed 2021-04-28 by DurandA

enjoy-digital, 2021-04-28:
@DurandA: No worries.