enjoy-digital/litex #909: Option to keep the CPU in reset state

jedrzejboczar, 2021-05-13:
It is currently possible to generate the SoC with `integrated_rom_mode = "rw"` and then reload the BIOS during execution, which is really helpful when debugging something that only needs changes in the BIOS firmware itself. But rewriting the memory while the CPU is executing from it can lead to unexpected results. 

I sometimes experience random failures when uploading new BIOS via EtherBone and I suspect that the CPU may sometimes stall the bus preventing the upload from completing, but this is just my guess (I get a timeout in litex_server and have to reupload the bitstream).

Currently there is no way to keep the CPU in the reset state for the whole upload time because a write to the CSR [only triggers the reset](https://github.com/enjoy-digital/litex/blob/53982acd9f448f365da337b0e2f3eb2db7428f73/litex/soc/integration/soc.py#L675). I understand that this CSR has to work that way or else the CPU would not be able to reset the SoC (would never go out of reset). Could there be another CSR that would allow for such behaviour? Currently I use the following local patch:
```diff
diff --git a/litex/soc/integration/soc.py b/litex/soc/integration/soc.py
index 8f5faec1..9e797bc6 100644
--- a/litex/soc/integration/soc.py
+++ b/litex/soc/integration/soc.py
@@ -659,6 +659,7 @@ class SoCController(Module, AutoCSR):
     def __init__(self, with_reset=True, with_scratch=True, with_errors=True):
         if with_reset:
             self._reset = CSRStorage(1, description="""Any write to this register will reset the SoC.""")
+            self._reset_hold = CSRStorage(1, description="""SoC will be in reset state as long as this register is 1.""")
         if with_scratch:
             self._scratch = CSRStorage(32, reset=0x12345678, description="""
                 Use this register as a scratch space to verify that software read/write accesses
@@ -672,7 +673,7 @@ class SoCController(Module, AutoCSR):
         # Reset
         if with_reset:
             self.reset = Signal()
-            self.comb += self.reset.eq(self._reset.re)
+            self.comb += self.reset.eq(self._reset.re | self._reset_hold.storage)

         # Errors
         if with_errors:
``` 
and then load the BIOS with:
```python
print('Reseting CPU')
wb.regs.ctrl_reset_hold.write(1)

print(f"Loading BIOS from: {bios_bin} starting at 0x{wb.mems.rom.base:08x} ...")
memwrite(wb, rom_data, base=wb.mems.rom.base)
wb.read(wb.mems.rom.base)

print('Rebooting CPU')
wb.regs.ctrl_reset_hold.write(0)
```
which avoids the random issues I had previously.

enjoy-digital, 2021-05-17:
Thanks @jedrzejboczar for reporting and the initial implementation, I also probably hit this issue and this would indeed be cleaner to stop the CPU when updating the ROM.  Unfortunately, your implementation will not work on most of the designs where we use the `reset` signal from `SoCController` to also reset the `CRG` (which would otherwise break your SoC bridge which does not seem to be the case). I would also like to avoid an extra CSR since will move the `SoCController` mapping. With https://github.com/enjoy-digital/litex/commit/d0e8de077c7e3bff9dd4bbb4a8c683009550d264 I added a separate `CSRField` to be able to control the SoC/CPU resets independently, this should cover this use-case, can you try it?

jedrzejboczar, 2021-05-17:
Thanks! From my initial tests it seems to work as expected, using:
```python
wb.regs.ctrl_reset.write(0b10)  # cpu_rst
# ...
wb.regs.ctrl_reset.write(0)
```

One thing I noticed when testing "manually" is that if I input some characters via UART while CPU is being held in reset, and then I put it out of reset, it reaches the BIOS console, but I cannot input anything (nothing shows up in the console) and I have to reload the bitstream. Not sure why this happens, but that shouldn't be any issue as I cannot see any reason to input anything while CPU is in reset.

closed 2021-05-17 by jedrzejboczar

enjoy-digital, 2021-05-17:
@jedrzejboczar: Thanks for the feedback. Is there anything specific with your UART (running in a different clock domain than the CPU?). Would you mind opening an issue for this? I think I also hit it a few weeks ago and could investigate.