enjoy-digital/litex #919: Building on windows fails - make inaccessible

SLongofono, 2021-05-17:
I am having issues building the xilinx_vc707 target on a fresh install using Windows 10.

This is using the latest Python3.9.  litex_setup appears to have run without incident for both the init install and gcc steps.

Per the quickstart and #728, I tried adding the env variable SHELL, both to the typical windows environment and the Makefile at litex/soc/software/libcompiler_rt/Makefile.  Powershell reports that cmd.exe is indeed in the path linked by SHELL:
```
PS C:\Users\PERCS\Desktop\litex\litex-boards\litex_boards\targets> echo $env:SHELL
C:\WINDOWS\system32\cmd.exe
```
Here is the full input and output I am seeing.  I can also attach a copy of the generated build directory in case that is helpful.

```
PS C:\Users\PERCS\Desktop\litex\litex-boards\litex_boards\targets> python .\xilinx_vc707.py --build
INFO:SoC:        __   _ __      _  __
INFO:SoC:       / /  (_) /____ | |/_/
INFO:SoC:      / /__/ / __/ -_)>  <
INFO:SoC:     /____/_/\__/\__/_/|_|
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2021-05-17 14:18:15)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7vx485tffg1761-2.
INFO:SoC:System clock: 125.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:S7MMCM:Creating S7MMCM, speedgrade -2.
INFO:S7MMCM:Registering Differential ClkIn of 200.00MHz.
INFO:S7MMCM:Creating ClkOut0 sys of 125.00MHz (+-10000.00ppm).
INFO:S7MMCM:Creating ClkOut1 sys4x of 500.00MHz (+-10000.00ppm).
INFO:S7MMCM:Creating ClkOut2 idelay of 200.00MHz (+-10000.00ppm).
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x20000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:S7MMCM:Config:
divclk_divide : 1
clkout0_freq  : 125.00MHz
clkout0_divide: 8
clkout0_phase : 0.00°
clkout2_freq  : 200.00MHz
clkout2_divide: 5
clkout2_phase : 0.00°
clkout1_freq  : 500.00MHz
clkout1_divide: 2
[... 171 lines cut ...]
- uart   : 0
- timer0 : 1
INFO:SoC:--------------------------------------------------------------------------------
Attempting to invoke make with command list: ['make', '-C', 'C:\\Users\\PERCS\\Desktop\\litex\\litex-boards\\litex_boards\\targets\\build\\xilinx_vc707\\software\\libcompiler_rt', '-f', 'c:\\users\\percs\\desktop\\litex\\litex\\litex\\soc\\software\\libcompiler_rt\\Makefile']
'make' is not recognized as an internal or external command,
operable program or batch file.
Traceback (most recent call last):
  File "C:\Users\PERCS\Desktop\litex\litex-boards\litex_boards\targets\xilinx_vc707.py", line 116, in <module>
    main()
  File "C:\Users\PERCS\Desktop\litex\litex-boards\litex_boards\targets\xilinx_vc707.py", line 106, in main
    builder.build(run=args.build)
  File "c:\users\percs\desktop\litex\litex\litex\soc\integration\builder.py", line 273, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "c:\users\percs\desktop\litex\litex\litex\soc\integration\builder.py", line 233, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "C:\Program Files\Python39\lib\subprocess.py", line 368, in check_call
    retcode = call(*popenargs, **kwargs)
  File "C:\Program Files\Python39\lib\subprocess.py", line 349, in call
    with Popen(*popenargs, **kwargs) as p:
  File "C:\Program Files\Python39\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\Python39\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

nickoe, 2021-05-17:
Will it work if you install make on windows and make sure it is available on the path?

SLongofono, 2021-05-17:
@nickoe I was assuming that running the setup script ( or later running it with the gcc argument) did as much, otherwise I would expect make/cygwin/mingw to be listed among the dependencies for quickstart.  I'll try one to see if I can get further and edit with the result here

nickoe, 2021-05-18:
@SLongofono I am not the maintainer of litex, just a user trying to get the hang of it. I have not tried to use it on windows directly, only via wsl2, which is not in the windows environment.. soo.

But I would rather suggest a way to actually make the error more user friendly, and possibly update the `litex_setup.py` script to possibly use chocolately directly or some other way to fetch a local `make.exe` for windows?

SLongofono, 2021-05-23:
Using chocolately would be slightly more involved, but it worked for the interim.  Since the script is already pulling various tools to build riscv and gcc, maybe 
[GnuWin](https://downloads.sourceforge.net/project/gnuwin32/make/3.81/make-3.81.exe?ts=gAAAAABgqpR37Tv3s615-HKFZyPt409eGjwS5X6QRu-rTLA0K1qcY0Nre6Ti9oZ7Y-NSAJdBjvHG9zWJmPQ1Tt7rbDWF3VueUA%3D%3D&r=https%3A%2F%2Fsourceforge.net%2Fprojects%2Fgnuwin32%2Ffiles%2Fmake%2F3.81%2Fmake-3.81.exe%2Fdownload%3Fuse_mirror%3Diweb%26download%3D) would be a better approach?

It makes sense to me to fetch this at the end of the repository updates, and then run the setup which includes building the GnuWin dependencies.

enjoy-digital, 2021-05-25:
Thanks @SLongofono for the feedback. I'll look at it soon and will make sure we list this as a dependency or provide an easy way to install it.

enjoy-digital, 2021-06-17:
make/GnuWin dependency has been added to the LiteX Wiki installation page.

closed 2021-06-17 by enjoy-digital