enjoy-digital/litex #921: Architecture exploration

navaneeth-cirel, 2021-05-21:
Hi,

I am new to litex and it really looks interesting to me for an SoC design. It would be great if someone can clear a few doubts of mine.

1. Is it possible to perform cycle accurate simulation in litex environment ? i.e to determine the number of cycles consumed in completing an instruction.
2. How to analyze/evaluate the performance of an architecture, for e.g. determine the bus utilization, find the optimum operating frequency for CPU and buses etc.

Thanks
Navaneeth

enjoy-digital, 2021-05-25:
Hi @navaneeth-cirel,

thanks for your interest in the project. It is indeed possible to perform cycle accurate simulation by using [litex_sim](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_sim.py) simulation as a basis and extend it to fit your needs.

With LiteX installed (https://github.com/enjoy-digital/litex/wiki/Installation), you'll be able to run a simple simulation by just running `litex_sim` command. You can then customize the cpu with `--cpu-type` argument, enable SDRAM with `--with-sdram`, etc...  (`--help` will provide you the list of supported arguments).

It's also possible to generate Waveforms with `--trace` or insert `Display` traces in your code (similar to C printf but for `HDL` traces/debug) and create specific debug logic that will do the measurements for you.

Using and extending similar principles, Antmicro built an automatic CPU benchmark in https://github.com/antmicro/embench-tester with the results published here https://antmicro.github.io/embench-tester/ that will probably be interesting for you since close to your needs.

navaneeth-cirel, 2021-06-01:
Hi @**enjoy-digital** ,

Thank you for your prompt and detailed response. 
Just to understand, does litex_sim use the inbuilt simulator from Migen or Verilator?

navaneeth-cirel, 2021-06-09:
> Hi @navaneeth-cirel,
>
> thanks for your interest in the project. It is indeed possible to perform cycle accurate simulation by using [...]

Hi @enjoy-digital ,

From your references I was able to setup a simple SoC to perform simulation and view the waveforms for debug. Thank you for that. 

I have a few doubts: 

1. Is there any sample code or wiki on how to use `Display` ,as you mentioned, in our code.
2. Is there a way to start the waveform dump through the C code which CPU executes? I know we can use `--trace-start` and `--trace-end` to specify when to start the waveform dump. My context here is, if I want to debug a custom peripheral which I will design and add into the SoC, I can enable the waveform dump through my C code when that peripheral is going to be used.

Thanks,
Navaneeth

enjoy-digital, 2021-06-17:
Hi @navaneeth-cirel,

thanks for the feedback,

Here is an example of use for Display:

```python3
        cycles = Signal(32)
        self.sync += cycles.eq(cycles + 1)
        self.sync += If(cycles == 1000000,
            Display("-"*80),
            Display("Cycles: %d", cycles),
            Finish(),
        )
```

It's also indeed possible to start tracing from software. To test this, you can use `--sim-debug` with `litex_sim` that will enable tracing control from software: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_sim.py#L309
 and will add a CSR to the SoC to allow software control: https://github.com/enjoy-digital/litex/blob/9bec0ce7a28afb498c7b7eb79943c270225c061a/litex/build/sim/platform.py#L67-L73

enjoy-digital, 2021-06-17:
@navaneeth-cirel: I'm closing since I think the questions are answered, but feel free to re-open if not.

closed 2021-06-17 by enjoy-digital