enjoy-digital/litex #922: dram initialization fails on ecp5 boards

gsomlo, 2021-05-25:
ECP5 bitstream for either `trellisboard` or `lambdaconcept_ecpix5` built using current-as-of-this-morning toolchain (yosys df2b79c, prjtrellis fe1c39c, and nextpnr e19d44e) consistently fails to pass DRAM initialization (see further below).

Building with the following command line:
```
litex-boards/litex_boards/targets/lambdaconcept_ecpix5.py --build \\
      --cpu-type vexriscv --cpu-variant linux --sys-clk-freq 50e6 \\
      --with-ethernet --with-sdcard
```

Console log during boot:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on May 24 2021 21:00:26
 BIOS CRC passed (cc15ed6a)

 Migen git sha1: --------
 LiteX git sha1: 5e40709c

--=============== SoC ==================--
CPU:            VexRiscv_Linux @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          524288KiB 16-bit @ 200MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |11111001| delays: 02+-02
  m0, b01: |00011001| delays: 04+-01
  m0, b02: |00011001| delays: 04+-01
  m0, b03: |00011001| delays: 04+-01
  best: m0, b00 delays: 02+-02
  m1, b00: |11111111| delays: 04+-04
  m1, b01: |00011111| delays: 05+-02
  m1, b02: |00011111| delays: 05+-02
  m1, b03: |00011111| delays: 05+-02
  best: m1, b00 delays: 04+-04
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  64/256
  addr errors: 2080/8192
  data errors: 524282/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> 
```

Also getting the same results when using `--cpu-type rocket` (which is how I noticed the problem to begin with). This used to work in the past, but neither `litex` nor the `yosys/trellis/nextpnr` combo are easily bisect-able, so I'm hoping either @enjoy-digital 
or @gatecat will have a good hunch to start us off in the right direction... :)

PS. I also tried building with vexriscv at `--sys-clk-freq 75e6` before submitting, and still getting a failure.

enjoy-digital, 2021-05-25:
Thanks @gsomlo for reporting, I'm looking at this. I've already been able to reproduce and see that it was related to some `liblitedram` changes that we did recently to improve DDR4/LPDDR4 calibration (so is not related to Yosys/Nextpnr), I'm looking further.

enjoy-digital, 2021-05-25:
This is fixed with https://github.com/enjoy-digital/litex/commit/d3560e57729493320fe28c7334bebaffbea73d22.

closed 2021-05-25 by enjoy-digital