enjoy-digital/litex #925: Memtest fails on ecpix5

chrivers, 2021-05-28:
Hello LiteX

I have a couple of Ecpix5 boards from Lambda Concept.

I've been trying to build litex for these boards, and see if the ddr3 ram controller works.

However, it does not.

I've enabled some additional debug output, and recompiled the bitstream. Please see the following output:

```
 Migen git sha1: 3ffd64c
 LiteX git sha1: 55344b4c

--=============== SoC ==================--
CPU:            VexRiscv @ 75MHz
BUS:            AXI-LITE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          524288KiB 16-bit @ 300MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Read leveling:
  m0, b00: |........| delays: 04+-04
  m0, b01: |.111....| delays: 00+-00
  m0, b02: |........| delays: 00+-00
  m0, b03: |........| delays: 00+-00
  best: m0, b00 delays: 04+-04
  m1, b00: |......43| delays: 03+-03
  m1, b01: |.111..33| delays: 00+-00
  m1, b02: |.111..22| delays: 00+-00
  m1, b03: |.111..22| delays: 00+-00
  best: m1, b00 delays: 03+-03
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
  bus errors:  32/256
  addr errors: 7168/8192
  data errors: 458743/524288
Memtest KO
Memory initialization failed

--============= Console ================--
```

I've tried using `ddr3_mr_gen.py` to test a large number of different `sdram_mr_write` variations, and recompiled the bitstream for 75MHz, 50MHz, 60MHz, 80MHz, and 90MHz.

Some of these combinations show a slight change in the initialization output, but not a single one can get past the memtest.

Here are some examples of problems I've caught debug logs for (not necessarily 100% same config as the above output)

Sometimes, some bits are all ones?
```
memtest_bus error @ 0x4000001c: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000028: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000034: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000040: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x4000009c: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400000a8: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400000b4: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400000c0: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x4000011c: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000128: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000134: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x40000140: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x4000019c: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400001a8: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400001b4: 0xaaffaaaa vs 0xaaaaaaaa
memtest_bus error @ 0x400001c0: 0xaaffaaaa vs 0xaaaaaaaa
```

Sometimes, some bits are flipped?
```
memtest_data error @ 0x4000070c: 0x4efefa39 vs 0x4e70fad5
memtest_data error @ 0x40000710: 0xa7a87daf vs 0xa7187d69
memtest_data error @ 0x40000714: 0xd3073ea6 vs 0xd3ac3eb7
memtest_data error @ 0x40000718: 0xe9831f53 vs 0xe9f61f58
memtest_data error @ 0x4000071c: 0x74e10faa vs 0x74fb0fac
memtest_data error @ 0x40000720: 0x3a188769 vs 0x3a7d87d6
memtest_data error @ 0x40000724: 0x1dacc3b7 vs 0x1d3ec3eb
```

```
memtest_addr error @ 0x4000007c: 0x0000001a vs 0x0000001f
memtest_addr error @ 0x40000080: 0x0000001b vs 0x00000020
memtest_addr error @ 0x40000084: 0x0000001c vs 0x00000021
memtest_addr error @ 0x40000088: 0x00000025 vs 0x00000022
memtest_addr error @ 0x4000008c: 0x0000001e vs 0x00000023
memtest_addr error @ 0x40000090: 0x0000001f vs 0x00000024
memtest_addr error @ 0x40000094: 0x00000020 vs 0x00000025
memtest_addr error @ 0x40000098: 0x00000029 vs 0x00000026
memtest_addr error @ 0x4000009c: 0x00000022 vs 0x00000027
memtest_addr error @ 0x400000a0: 0x00000023 vs 0x00000028
memtest_addr error @ 0x400000a4: 0x00000024 vs 0x00000029
```

enjoy-digital, 2021-06-01:
Hi @chrivers,

thanks for the feedback, there has been a regression recently on ECP5 calibration that has been fixed with https://github.com/enjoy-digital/litex/commit/d3560e57729493320fe28c7334bebaffbea73d22, can you make sure you are using an upstream version with this change?

Otherwise, here is a bitstream I just generated and tested on the ECPIX-5 with the default parameters, that works on my hardware, can you test it on yours? 
[lambdaconcept_ecpix5_2021_06_01.zip](https://github.com/enjoy-digital/litex/files/6574512/lambdaconcept_ecpix5_2021_06_01.zip)

enjoy-digital, 2021-06-17:
Closing since was probably related to the regression identified (https://github.com/enjoy-digital/litex/commit/d3560e57729493320fe28c7334bebaffbea73d22) but please re-open @chrivers if you still have troubles when testing.

closed 2021-06-17 by enjoy-digital

chrivers, 2021-09-15:
Hello again! Sorry, I completely dropped the ball on this.

I tried your custom bit stream (thank you very much for making that), and it works perfectly!

Just to confirm: commit d3560e5 also fixed it for me (assuming that was the primary contents of the custom bitstream).

Thanks for the help :)

enjoy-digital, 2021-09-15:
Great, thanks for the feedback.