enjoy-digital/litex #928: DE10lite stuck in memtest

Whiteck, 2021-06-03:
Hello, 

I have an issue with DE10 lite.
I try to implement SERV on it. But, that don't work so I try a more supported core, Vexriscv. 
The problem still the same, the memtest failed everytimes : 
``` bash
__   _ __      _  __

/ /  (_) /____ | |/_/

/ /__/ / __/ -_)>  <

/____/_/\__/\__/_/|_|

Build your hardware, easily!

(c) Copyright 2012-2021 Enjoy-Digital

(c) Copyright 2007-2015 M-Labs

BIOS built on Jun  3 2021 13:39:09

BIOS CRC passed (b9ed3c75)

Migen git sha1: 3ffd64c

LiteX git sha1: a064e9d0

--=============== SoC ==================--

CPU:		VexRiscv @ 50MHz

BUS:		WISHBONE 32-bit @ 4GiB

CSR:		32-bit data

ROM:		128KiB

SRAM:		8KiB

L2:		8KiB

SDRAM:		65536KiB 16-bit @ 50MT/s (CL-2 CWL-2)

--========== Initialization ============--

Initializing SDRAM @0x40000000...

Switching SDRAM to software control.

Switching SDRAM to hardware control.

Memtest at 0x40000000 (2.0MiB)...

Write: 0x40000000-0x40000000 0B
Write: 0x40000000-0x40020000 128.0KiB
Write: 0x40000000-0x40040000 256.0KiB
Write: 0x40000000-0x40060000 384.0KiB
Write: 0x40000000-0x40080000 512.0KiB
Write: 0x40000000-0x400a0000 640.0KiB
Write: 0x40000000-0x400c0000 768.0KiB
Write: 0x40000000-0x400e0000 896.0KiB
Write: 0x40000000-0x40100000 1.0MiB
Write: 0x40000000-0x40120000 1.1MiB
Write: 0x40000000-0x40140000 1.2MiB
Write: 0x40000000-0x40160000 1.3MiB
Write: 0x40000000-0x40180000 1.5MiB
Write: 0x40000000-0x401a0000 1.6MiB
Write: 0x40000000-0x401c0000 1.7MiB
Write: 0x40000000-0x401e0000 1.8MiB
Write: 0x40000000-0x40200000 2.0MiB

Read: 0x40000000-0x40000000 0B
Read: 0x40000000-0x40020000 128.0KiB
Read: 0x40000000-0x40040000 256.0KiB
Read: 0x40000000-0x40060000 384.0KiB
Read: 0x40000000-0x40080000 512.0KiB
Read: 0x40000000-0x400a0000 640.0KiB
Read: 0x40000000-0x400c0000 768.0KiB
Read: 0x40000000-0x400e0000 896.0KiB
Read: 0x40000000-0x40100000 1.0MiB
Read: 0x40000000-0x40120000 1.1MiB
Read: 0x40000000-0x40140000 1.2MiB
Read: 0x40000000-0x40160000 1.3MiB
Read: 0x40000000-0x40180000 1.5MiB
Read: 0x40000000-0x401a0000 1.6MiB
Read: 0x40000000-0x401c0000 1.7MiB
Read: 0x40000000-0x401e0000 1.8MiB
Read: 0x40000000-0x40200000 2.0MiB

bus errors:  256/256

addr errors: 8191/8192

data errors: 524288/524288

Memtest KO

Memory initialization failed
``` 
At the beginning, I tough it was an problem with SDRAM but without, the program seems to not be implement.  

Also, another problem, when some command are send trough the UART, no answer are coming back. I try some "memory_read", "memory_write" and "reboot".  

Do you have some ideas to make the core works ?

enjoy-digital, 2021-06-06:
Hi @Whiteck,

there is probably something wrong in the De10Lite support (I remember other users had issue in the past also with the memtest). I unfortunatelly don't have access to a De10Lite to investigate. Do you eventually have another board you could use until we can fix the De10Lite support?

Whiteck, 2021-06-07:
Hi @enjoy-digital,

Unfortunately, I have just this board. One of the goals to my project it make de10lite works, so I have to continue to investigate and find what's happen.

enjoy-digital, 2021-06-07:
IIRC it was working at some point but something broke it. To ease debug, you can un-comment the #defines here to enable memtest debug: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libbase/memtest.c#L10-L12

You can also try to change the phase of the sys_ps clk here: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/terasic_de10lite.py#L48 to 45 or 180 and see if you notice differences.

enjoy-digital, 2021-06-07:
@Whiteck: It could also be interesting to test with this previous version of the CRG that I think has been tested succesfully in the past: 
```

# CRG ----------------------------------------------------------------------------------------------

class _CRG(Module):
    def __init__(self, platform):
        self.clock_domains.cd_sys    = ClockDomain()
        self.clock_domains.cd_sys_ps = ClockDomain()
        self.clock_domains.cd_vga    = ClockDomain(reset_less=True)

        # # #

        # main input clock for PLL
        clk50 = platform.request("clk50")

        # sys clk / sdram clk  / vga_clk from PLL
        pll_locked  = Signal()
        pll_clk_out = Signal(6)
        self.specials += \
            Instance("ALTPLL",
                p_BANDWIDTH_TYPE         = "AUTO",
                p_CLK0_DIVIDE_BY         = 1,
                p_CLK0_DUTY_CYCLE        = 50,
                p_CLK0_MULTIPLY_BY       = 1,
                p_CLK0_PHASE_SHIFT       = "0",
                p_CLK1_DIVIDE_BY         = 1,
                p_CLK1_DUTY_CYCLE        = 50,
                p_CLK1_MULTIPLY_BY       = 1,
                p_CLK1_PHASE_SHIFT       = "-10000",
                p_CLK2_DIVIDE_BY         = 2,
                p_CLK2_DUTY_CYCLE        = 50,
                p_CLK2_MULTIPLY_BY       = 1,
                p_CLK2_PHASE_SHIFT       = "0",
                p_COMPENSATE_CLOCK       = "CLK0",
                p_INCLK0_INPUT_FREQUENCY = 20000,
                p_INTENDED_DEVICE_FAMILY = "MAX 10",
                p_LPM_TYPE               = "altpll",
                p_OPERATION_MODE         = "NORMAL",
                i_INCLK                  = clk50,
                o_CLK                    = pll_clk_out,
                i_CLKENA                 = 0x3f,
                i_EXTCLKENA              = 0xf,
                i_FBIN                   = 1,
                i_PFDENA                 = 1,
                i_PLLENA                 = 1,
                o_LOCKED                 = pll_locked,
            )
        self.comb += [
            self.cd_sys.clk.eq(pll_clk_out[0]),
            self.cd_sys_ps.clk.eq(pll_clk_out[1]),
            self.cd_vga.clk.eq(pll_clk_out[2])
        ]
        self.specials += [
            AsyncResetSynchronizer(self.cd_sys,    ~pll_locked),
            AsyncResetSynchronizer(self.cd_sys_ps, ~pll_locked),
            AsyncResetSynchronizer(self.cd_vga,    ~pll_locked),
        ]
        self.comb += platform.request("sdram_clock").eq(self.cd_sys_ps.clk)

```

Whiteck, 2021-06-08:
Hello, 

So, both solutions doesn't work, with memtest_debug, I just found : 0xffffffff vs 0xaaaaaaaa.
Data aren't write in SDRAM, after the flush, they don't exist anymore and create all the errors.
``` bash 
data before flush l2 

val 1 : 0xaaaaaaaa 

val 64 : 0xaaaaaaaa 

val 128 : 0xaaaaaaaa 

data after flush l2 

val 1 : 0x0 

val 64 : 0x0 

val 128 : 0x0 
```
Do you know if some update to sdram code in python can create this problem ? (specially gensdrphy code)

and other problem, when some instructions are send to the soc by uart, the soc respond rarely. 

Do you know, how I can debug the serial port ?

enjoy-digital, 2021-06-08:
@Whiteck: This could indeed be interesting to use some old versions of the code to see the behaviour and understand what exactly broke the DE10Lite support. I think this commit was working: https://github.com/litex-hub/litex-boards/commit/9d6a6c1bcb0ebd777a99104b5ea060f20e403ecf. This could be interesting to revert all the dependencies to the date of this commit (at least LiteX-Boards/LiteX/LiteDRAM) and see if it behaves correctly. If so, you could progressively update the dependencies and if you identify the commit breaking things, it will probably be easy to understand the issue.

Whiteck, 2021-06-30:
Hi @enjoy-digital ,

So for the SDRAM issue, I don't find it. To make DE10 works, I add an intern main-ram (we can go up to 128KiB). 
For the uart issue, this came from my USB to UART.

enjoy-digital, 2021-06-30:
Thanks @Whiteck for the feedback, sorry I don't have the hardware so can't help that much on this yet. (I'll maybe order it in the future).

esourdin, 2021-07-26:
Dear @enjoy-digital,

So, I rework on the issue and with the help of a ​de1soc, I found what happen. 
On a DE10lite, the sdram isn't use by the board. If I take an oscillo and check pin clk/cas/ras/en, no signal can be found. 

So, when I compare de1soc and de10lite projet on quartus,  4 files are missing for the de10.
These files are : "altsyncram:storage_2_rtl_0" to "altsyncram:storage_5_rtl_0". In theses files, the sdram fsm are instanciate. 

I can reproduce that issue on a de1soc, if a put a value to "integrated_main_ram_size". 

Do you have an idea for fixing this issue ?

enjoy-digital, 2021-07-26:
@esourdin: If you build the default de10lite configuration and don't have a valid clock, there is indeed something wrong, this is generated here: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/terasic_de10lite.py#L52 You can maybe try to duplicate this to a free IO to check.

esourdin, 2021-07-28:
@enjoy-digital, the clock are good (I've check it on oscillo and compare with de1soc clock). 

But, I've find an error, but I don't really understand what I have to change to avoid it. This is : 

{ { "Info" "IINFER_RAM_UNINFERRED_DUE_TO_SIZE" "storage_5 " "RAM logic \"storage_5\" is uninferred due to inappropriate RAM size" {  } { { "terasic_de10lite.v" "storage_5" { Text "/home/esourdin/liteX/litex-boards/litex_boards/targets/build/terasic_de10lite/gateware/terasic_de10lite.v" 5437 -1 0 } }  } 0 276004 "RAM logic \"%1!s!\" is uninferred due to inappropriate RAM size" 0 0 "Design Software" 0 -1 1627462624126 ""}

Do you know, what can I try to avoid it ?

closed 2021-09-07 by esourdin