enjoy-digital/litex #929: How to generate "case" "endcase" verilog statements with python

jimmymagemtek, 2021-06-05:
Hi Sir,
As title, I need to implement something like FSM.
And I can't use FSM(), because I found the assignment statement is always assigned as zero in the beginning of the loop with COMB logic.

And there seems no such statement like "If(...     ).Elseif()" available.    

Please advise. Thanks.

Jimmy

gsomlo, 2021-06-05:
Hi Jimmy,

On Sat, Jun 05, 2021 at 07:18:22AM -0700, jimmymagemtek wrote:
> And there seems no such statement like "If(... ).Elseif()" available.

Your question is probably better suited for
Migen (https://github.com/m-labs/migen, or
https://m-labs.hk/gateware/migen), rather
than Litex (which happens to be *written*
in Migen).

As for "If().Elseif()", you're very close:
it's "Elif()" instead:

https://m-labs.hk/migen/manual/fhdl.html#if

There's a Migen tutorial linked from
https://m-labs.hk/gateware/migen, which does
(IIRC) illustrate how Verilog is generated
from various Migen statements.

HTH,
--Gabriel

jimmymagemtek, 2021-06-05:
Hi Gabriel,
Thanks. I tried ".Elif", but it still can't generate "case" "endcase" structure.
The reason I need it because "FSM()" is translated into "case" and simulation works properly. But with "If .. else" verilog code, simulation stop since it seems to be unable to change the variable, for unknown reason.

Anyway. Thanks.

Jimmy

enjoy-digital, 2021-06-17:
Hi @jimmymagemtek,

In a FSM, the signals will indeed use the reset value when not assigned, which is generally 0. But it's possible to use a different reset value: `my_signal= Signal(1, reset=1`).

Migen also has `Case` statement support, ex:
https://m-labs.hk/migen/manual/fhdl.html#case
https://github.com/enjoy-digital/litex/blob/5061a368daa9028c7720f10aec25286e09687099/litex/soc/interconnect/stream.py#L279-L295

closed 2021-06-17 by enjoy-digital