enjoy-digital/litex #933: Memory test fails on arty7 + Rocket

developandplay, 2021-06-07:
I wanted to try version a064e9d of Litex with Rocket.
Unfortunately it failed the memory tests.
I went back a bit further and also tried 21273ffe which failed them as well.
On the ulx3s the initialization works without any issues.

Additionally I tried vexriscv at 21273ffe which passes the memory tests on both arty7 and ulx3s.

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  8 2021 01:11:38
 BIOS CRC passed (49f9001e)

 Migen git sha1: 3ffd64c
 LiteX git sha1: 21273ffe

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
SDRAM:          262144KiB 16-bit @ 400MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (2MiB)...
  Write: 0x80000000-0x80200000 2MiB     
   Read: 0x80000000-0x80200000 2MiB     
  bus errors:  0/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--
```

developandplay, 2021-06-08:
VexRiscv was tested at 100Mhz.

I now tested VexRiscv at 50Mhz and also encounter memory errors.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  8 2021 15:11:29
 BIOS CRC passed (2c69bae1)

 Migen git sha1: 3ffd64c
 LiteX git sha1: bd146351

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          262144KiB 16-bit @ 400MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  80/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--
```

developandplay, 2021-06-08:
With the workaround from litex-hub/litex-boards#166 the memory test now succeeds using VexRiscV@50Mhz on the arty board.

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  8 2021 15:29:56
 BIOS CRC passed (c052ac32)

 Migen git sha1: 3ffd64c
 LiteX git sha1: bd146351

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          262144KiB 16-bit @ 400MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111111111111111111111000| delays: 14+-14
  m0, b02: |00000000000000000000000000000001| delays: 31+-00
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 14+-14
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11111111111111111111111111111000| delays: 14+-14
  m1, b02: |00000000000000000000000000000001| delays: 31+-01
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 14+-14
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (2.0MiB)...
  Write speed: 15.8MiB/s
   Read speed: 13.0MiB/s

--============== Boot ==================--
```

developandplay, 2021-06-08:
With the workaround from litex-hub/litex-boards#166 the memory test now succeeds using Rocket@50Mhz on the arty board.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun  8 2021 16:06:43
 BIOS CRC passed (8a1b0c8a)

 Migen git sha1: 3ffd64c
 LiteX git sha1: bd146351

--=============== SoC ==================--
CPU:            RocketRV64[imac] @ 50MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
SDRAM:          262144KiB 16-bit @ 400MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111111111111111111111000| delays: 14+-14
  m0, b02: |00000000000000000000000000000001| delays: 31+-00
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 14+-14
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11111111111111111111111111111000| delays: 14+-14
  m1, b02: |00000000000000000000000000000011| delays: 31+-01
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 14+-14
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x00000080000000 (2.0MiB)...
  Write speed: 22.4MiB/s
   Read speed: 25.7MiB/s

--============== Boot ==================--
```

developandplay, 2021-06-08:
So what do we make from this now? :thinking:

enjoy-digital, 2021-06-08:
@developandplay: Thanks for the tests, now we know a little bit more what's going on. I'll try to improve this soon.

developandplay, 2021-06-08:
Just tested Rocket@100Mhz with the workaround and it also works.

enjoy-digital, 2021-06-08:
At 100MHz, it should work with Rocket without the workaround no?

developandplay, 2021-06-08:
Yes it did. Just wanted to add this workaround doesn't lead to regression at 100Mhz.

gsomlo, 2021-06-08:
On Tue, Jun 08, 2021 at 09:48:59AM -0700, developandplay wrote:
> Yes it did. Just wanted to add this workaround doesn't lead to regression at
> 100Mhz.

Just curious, if you build litex+rocket at 100MHz, does Vivado think
the design passes timing? I remember getting timing failures from
Vivado (on the nexys4ddr) at anything significantly higher than 50MHz
(admittedly, after adding both --with-ethernet and --with-sdcard).

developandplay, 2021-06-08:
Yep it doesn't meet timing constraints at 100Mhz.

```
------------------------------------------------------------------------------------------------
| Design Timing Summary
| ---------------------
------------------------------------------------------------------------------------------------

    WNS(ns)      TNS(ns)  TNS Failing Endpoints  TNS Total Endpoints      WHS(ns)      THS(ns)  THS Failing Endpoints  THS Total Endpoints     WPWS(ns)     TPWS(ns)  TPWS Failing Endpoints  TPWS Total Endpoints  
    -------      -------  ---------------------  -------------------      -------      -------  ---------------------  -------------------     --------     --------  ----------------------  --------------------  
     -4.396   -10587.749                   8080                35500        0.035        0.000                      0                35500        0.264        0.000                       0                 13216  

Timing constraints are not met.
```

developandplay, 2021-07-04:
Note for my future self and others using Blackparrot:
When reducing the speed of Blackparrot to 20Mhz to meet timing constraints with the FPU -> increasing CL further is necessary.
Using
`python3 ddr3_mr_gen.py --cl=8 --cwl=5`
which leads to:
```
Commands to be used with LiteX BIOS:
sdram_mr_write 0 2624
sdram_mr_write 1 2054
sdram_mr_write 2 512
```
the memory test succeeds again.

enjoy-digital, 2021-09-14:
This has been fixed with https://github.com/enjoy-digital/litedram/commit/6f323f6a7aa032c8342b5931f4a562d2e28969e9 and tested on Rocket at 50Mhz with it: 

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep 14 2021 16:31:47
 BIOS CRC passed (2f08fe54)

 Migen git sha1: a5bc262
 LiteX git sha1: 694878a3

--=============== SoC ==================--
CPU:		RocketRV64[imac] @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
SDRAM:		262144KiB 16-bit @ 400MT/s (CL-7 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111111111111111111000000| delays: 13+-13
  m0, b02: |00000000000000000000000000001111| delays: 30+-02
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 13+-13
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |11111111111111111111111111000000| delays: 13+-13
  m1, b02: |00000000000000000000000000001111| delays: 30+-02
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 13+-13
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
Memtest OK
Memspeed at 0x00000080000000 (Sequential, 2.0MiB)...
  Write speed: 10.3MiB/s
   Read speed: 16.3MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> 

```

closed 2021-09-14 by enjoy-digital