enjoy-digital/litex #935: Code upload to simulated target through lxterm fails

navaneeth-cirel, 2021-06-08:
Hi @enjoy-digital,

My aim is to be able to upload firmware to a simulated target through lxterm, so that I can try out various Risc-V code in my SoC design simulation. I have setup a sample simulated SoC and set 

`
sim_config.add_module("serial2tcp", "serial", args={"port" : 1111})
`

to give access to lxterm through TCP socket. I am running lxterm like below

`
lxterm --serial-boot --kernel cirel-soc/build/sim/demo/demo.bin socket://localhost:1111
`

to try and upload the binary of the [demo](https://github.com/enjoy-digital/litex/tree/master/litex/soc/software/demo) project.

However the upload fails at 88% of completion.

[image]

Please let me know if there's a better method to achieve the same or what I can try to solve this error.

developandplay, 2021-06-08:
You could try sideloading the binary directly to simulation which should also be quicker.
For that you can use`--ram-init cirel-soc/build/sim/demo/demo.bin` in `litex_sim`.

navaneeth-cirel, 2021-06-08:
@developandplay if I understand correctly this adds the binary file directly in the final Verilog file generated for the SoC, so wouldn’t this involve building the SoC every time I change the firmware binary ? Please correct me if I am wrong.
Though the build time is reasonably low now, I assume it will increase as the design becomes complex.

closed 2026-06-01 by enjoy-digital