enjoy-digital/litex #938: How does Litex expect 32bit reads/writes to be handled by 64bit CPUs?

developandplay, 2021-06-10:
Unfortunately I'm still stuck on https://github.com/enjoy-digital/litex/commit/4f82a36a when updating Blackparrot support to the most recent Litex version.

The current wishbone adapter in Blackparrot only handles 64bit reads/writes which means `sel` (corresponding to AXis `strb`) will always be 0xFF. If I do understand what is breaking https://github.com/enjoy-digital/litex/commit/4f82a36a for Blackparrot is the reliance on `sel` in https://github.com/enjoy-digital/litex/commit/75936775 which requires `sel` to be set to either 0x0F or 0xF0 when doing 32bit accesses.

However I'm not sure what Litex expects exactly from the CPUs Wishbone/AXI adapter. I also read the [discussion](https://github.com/enjoy-digital/litex/issues/544) behind https://github.com/enjoy-digital/litex/commit/4f82a36a and the [CSR bus wiki](https://github.com/enjoy-digital/litex/wiki/CSR-Bus) but when it comes to 64bit CPUs it doesn't seem very clear to me.

Maybe @gsomlo can chime as Rocket does support 32bit read/writes for CSR access and I saw he asked how Rocket handles 32bit read/writes [here](https://github.com/chipsalliance/rocket-chip/issues/2185).

gsomlo, 2021-06-10:
On Thu, Jun 10, 2021 at 05:19:07AM -0700, developandplay wrote:
> Maybe @gsomlo can chime as Rocket does support 32bit read/writes for
> CSR access and I saw he asked how Rocket handles 32bit read/writes here.

I remember having Rocket work with a 64-to-32-bit wishbone converter
at some point in the past. In recent memory, however, I've been
provisioning it with a dedicated point-to-point AXI link between
Rocket's dedicated cached-ram AXI port and LiteDRAM, without any
width conversion.

I'd recommend trying to build a LiteX+Rocket setup where the CPU
cached-ram AXI width (64 for the "linux" variant, 128 for "linuxd",
256 for "linuxq") does NOT match the board-specific LiteDRAM native
port, which would force a width conversion via Wishbone.

Does *that* still work (with Rocket)? I haven't tried recently, and
I'm stuck working in a place where I can't easily tinker with this
stuff, so I can't try it myself. But it would be good to know...

developandplay, 2021-06-10:
> I'd recommend trying to build a LiteX+Rocket setup where the CPU cached-ram AXI width (64 for the "linux" variant, 128 for "linuxd", 256 for "linuxq") does NOT match the board-specific LiteDRAM [...]

Yes it does. That is what is happening on the ulx3s with it's 16bit width memory bus.

I looked into this some more and I realised that blackparrot is still using the `periph_buses` like most 32bit cpus. Which I guess means it's wired up directly and doesn't use any conversion techniques from Litex.
https://github.com/enjoy-digital/litex/blob/0a932be4919ab7b6cdce05e77def94d4a1eedbbc/litex/soc/cores/cpu/blackparrot/core.py#L90

Additionally the conversion from  64-to-32-bit wishbone conversion seems to only be happening for AXI devices.
https://github.com/enjoy-digital/litex/blob/0a932be4919ab7b6cdce05e77def94d4a1eedbbc/litex/soc/integration/soc.py#L1300-L1330

So I guess a first step would then be to use the `memory_buses` in blackparrot.
Next adding the width conversion when using wishbone as a memory bus.

Does sound right or am I mixing up things here?

developandplay, 2021-06-11:
Just did a sanity check with 64bit backported to the latest Litex version.
As expected it passes the memory test so my assumption about this being the cause at least was correct.
Now I still need to figure out how to make blackparrot work with 32bit alignment though. :smile: :man_shrugging: 
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 11 2021 16:22:38
 BIOS CRC passed (2638c9f4)

 Migen git sha1: 3ffd64c
 LiteX git sha1: 0a932be4

--=============== SoC ==================--
CPU:            BlackParrotRV64[ima] @ 40MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:       32-bit data - 64-bit aligned
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 320MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
  bus errors:  0/256
  addr errors: 0/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> sdram_mr_write 0 2608
Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2054
Switching SDRAM to software control.
Writing 0x0806 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 512
Switching SDRAM to software control.
Writing 0x0200 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000001111111111111111111111111| delays: 20+-12
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 20+-12
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000011111111111111111111111111| delays: 19+-12
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 19+-12
Switching SDRAM to hardware control.

litex> sdram_test
Memtest at 0x00000080000000 (8.0MiB)...
  Write: 0x80000000-0x80800000 8.0MiB     
   Read: 0x80000000-0x80800000 8.0MiB     
Memtest OK

litex> 
```

developandplay, 2021-06-11:
@gsomlo The one thing that really confuses is the fact that on simulation everything works fine after switching to 32bit alignment. Even booting into linux works after adjusting risv-pk (of course in that case with `--sdram` off).

developandplay, 2021-06-12:
So what I'm trying to understand is why is it such that on Rocket removing CSR alignment does not affect the communication while on Blackparrot it does (at least on real hardware).

Quote by @enjoy-digital from #544:
> So we should just not change the current gateware regarding CSR data-width support: 8-bit and 32-bit and as tested with gsomlo on Rocket we can just remove the CSR alignment and always set it to 32-bit. This should already allow to simplify the software support.

gsomlo, 2021-06-12:
On Fri, Jun 11, 2021 at 06:45:35PM -0700, developandplay wrote:
> So what I'm trying to understand is why is it such that on Rocket removing CSR
> alignment does not affect the communication while on Blackparrot it does (at
> least on real hardware).

I'd defer to @enjoy-digital on the actual details, but this sounds
like maybe the appropriate strobe signals that should be driven by
the 64-bit CPU (indicating which 32-bit half of a 64-bit load/store
is actually desired) are getting "lost in translation".

enjoy-digital, 2021-06-14:
@developandplay: As  suggested by @gsomlo, this could indeed be related to the write strobe signals: Blackparrot could generate them incorrectly or LiteX handle them incorrectly. The best would be do a simulation and observe the accesses done by BlackParrot on the main bus and then on the CSR bus. (And compare this to Rocket).

developandplay, 2021-06-14:
@enjoy-digital Do you think this would be beneficial considering the memory issues only occur on my arty board?

As far as simulation goes I tried to get as close to my arty board as possible but there might still be some differences.
1. I used the `--with-sdram`, `--sdram-module`, `--sdram-data-width` parameters.
2. I extended the `MEMTEST_DATA_SIZE` to be the same in simulation as in hardware.
3. I turned of the litex l2 cache on the arty and in simulation. This leads to more memory errors on the arty but didn't lead to any memoy errors in simulation.

Are there any points I could be missing that prevent me from reproducing the errors in simulation?

developandplay, 2021-06-14:
@enjoy-digital 
Another option that I was thinking of - as neither Sadullah, you or me - got this 32bit alignment on the FPGA to work is whether we could reintroduce 64bit alignment...Limited to Blackparrot of course.

enjoy-digital, 2021-06-14:
@developandplay: The current issue is probably related to the CSR accesses that are done during the DDR3 calibration. These accesses are only done on hardware for the read leveling and skipped for the simulation, so that's probably the reason you don't reproduce the issue in simulation. Do you know why BlackParrot does not handle Write strobes?

developandplay, 2021-06-14:
I think Sadullahs project was more of a proof of concept where strobes were not necessary because everything was 64bit aligned. It's just with 4f82a36 that they suddenly were not anymore so I guess that's what lead to #641.

DDR3 calibration only happening on the actual hardware makes a lot of sense. That could be one reason why CSR alignment might not be an issue in simulation despite strb being set to just 0xFF.

I will look at a CSR access and compare between Rocket and Blackparrot. Any CSR read/write that might be interesting to look at because it triggers multiple different strobe configurations?

Another interesting thing is that with 64bit alignement backported to the latest Litex version, I don't even get into BIOS on simulation but on my arty it successfully passes the memory test.

enjoy-digital, 2021-06-14:
OK, at least now we understand what broke things, which is a good first step :) What's the granularity of the write accesses on the native BlackParrot interface? If it's 8-bit, I would recommend propagating that to the wishbone interface, this should fix the issue.

developandplay, 2021-06-14:
Yes it's 8-bit but isn't that the default granularity in Litex anyways?

enjoy-digital, 2021-06-14:
8-bit (a byte) is the standard and the default one in LiteX yes.

developandplay, 2021-06-14:
Ok I will give the setting select/strobe another shot then.
That was my starting point but then the memory errors in the test confused me as they don't obviously related to the CSRs.

developandplay, 2021-06-16:
I did succeed in setting up select/strobe properly.
Now however I'm getting some other differences between Rocket and Blackparrot.

**Rocket**
[image]

**Blackparrot**
[image]

This should be the same uart csr write request. Here Blackparrot inserts a second write request while on Rocket only one occurs. I'm not sure if that's a difference that should not matter or if I should investigate this further. As of now the memory errors are not resolved just by setting select/strobe...

enjoy-digital, 2021-06-16:
@developandplay: Good, the next step would probably be to add a large CSRStorage in your design (with a specific value) and make sure you are able to do read/write to it correctly and  eventually compare results with Rocket. Large CSR are not that common, but DRAM init use some for the DFI data, and if accesses on large CSR are not working correctly, it could be the issue. If you are not able to figure out the issue, you can also prepare PRs to allow me to test your current code, I could have a look.

developandplay, 2021-06-17:
Sound good I will look into that. Let me first see if I can figure it out on my own and hopefully gain some deeper understanding about CSRs along the way...
Is there a Litex core that uses the CSRStorage without too much added logic so I have somewhat of a reference?

enjoy-digital, 2021-06-17:
You can just use a simple module like this:
```python3
class DummyCSR(Module, AutoCSR):
    def __init__(self):
        self.my_csr = CSRStorage(128, init=0x12345678)
```
and in your SoC:
```python3
self.submodules.dummy_csr = DummyCSR()
```

developandplay, 2021-06-22:
Some simple test with the scratch register extended to a length of 64 and initialized with 0x1234567812345678 give me interesting results. (Even in simulation)
The first two reads and writes work just as expected.
The last one as well.
The third one seems to be problematic.
Doing a 32 bit write and then a read reveals that 32 bit LSB writes don't seem to work.

```

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 22 2021 22:27:29
 BIOS CRC passed (40b4d30a)

 Migen git sha1: 3ffd64c
 LiteX git sha1: b795f848

--=============== SoC ==================--
CPU:            BlackParrotRV64[ima] @ 1MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          65536KiB 32-bit @ 1MT/s (CL-2 CWL-2)

--========== Initialization ============--
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (8.0KiB)...
  Write: 0x80000000-0x80002000 8.0KiB   
   Read: 0x80000000-0x80002000 8.0KiB   
Memtest OK
Memspeed at 0x00000080000000 (8.0KiB)...
  Write speed: 1.8MiB/s
   Read speed: 1.8MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--

litex> mem_read 0x50000004 8
Memory dump:
0x50000004  78 56 34 12 78 56 34 12                          xV4.xV4.        

litex> mem_write 0x50000004 0x1111111111111111111 2

litex> mem_read 0x50000004 8                       
Memory dump:
0x50000004  11 11 11 11 11 11 11 11                          ........
       
litex> mem_write 0x50000004 0x12345678             

litex> mem_read 0x50000004 8          
Memory dump:
0x50000004  00 00 00 00 11 11 11 11                          ........    

litex> mem_write 0x50000008 0x12345678

litex> mem_read 0x50000004 8          
Memory dump:
0x50000004  00 00 00 00 78 56 34 12                          ....xV4. 
```

developandplay, 2021-06-22:
@enjoy-digital 
Thanks a lot for the help!!! This was quite a debugging journey but I finally figured out the 32bit alignment.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jun 22 2021 23:19:20
 BIOS CRC passed (ca188395)

 Migen git sha1: 3ffd64c
 LiteX git sha1: b795f848

--=============== SoC ==================--
CPU:            BlackParrotRV64[ima] @ 40MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            128KiB
SRAM:           8KiB
L2:             8KiB
SDRAM:          262144KiB 16-bit @ 320MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x80000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x00000080000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
  bus errors:  0/256
  addr errors: 32/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> sdram_mr_write 0 2608
Switching SDRAM to software control.
Writing 0x0a30 to MR0
Switching SDRAM to hardware control.

litex> sdram_mr_write 1 2054
Switching SDRAM to software control.
Writing 0x0806 to MR1
Switching SDRAM to hardware control.

litex> sdram_mr_write 2 512
Switching SDRAM to software control.
Writing 0x0200 to MR2
Switching SDRAM to hardware control.

litex> sdram_cal
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0 
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000001111111111111111111111111| delays: 19+-12
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 19+-12
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000001111111111111111111111111| delays: 19+-13
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 19+-12
Switching SDRAM to hardware control.

litex> sdram_test
Memtest at 0x00000080000000 (8.0MiB)...
  Write: 0x80000000-0x80800000 8.0MiB     
   Read: 0x80000000-0x80800000 8.0MiB     
Memtest OK

litex>
```

mithro, 2021-06-22:
@developandplay - Congratulations! Next stop, getting it working with SymbiFlow?

If you do want to take on the awesome / epic task of getting LiteX + BlackParrot + SymbiFlow going it would be awesome! It's likely to be very bumpy, but you'll be pushing forward the state of open source tooling in a pretty epic way!

Here are some tips;
  * Come join us on the #symbiflow IRC channel on https://libera.chat/ or [via Slack](https://join.slack.com/t/symbiflow/shared_invite/enQtNTkyMjcyNTkzOTY4LTU0MzhmYWNjOGMyMTkyNjA0MmEyMWM5OWY3ZDg5MWQ3ODlmOWQwZjk2YzBmMDBjMzkzMzNjYjkwYjAxZTMyNjQ).
 * SymbiFlow can already be used with LiteX and VexRISCV -- see the demo at https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html#linux-litex-demo).
 * You will need [Surelog](https://github.com/chipsalliance/Surelog) + [UHDM](https://github.com/chipsalliance/UHDM) for BlackParrot with the [Yosys integration](https://github.com/alainmarcel/uhdm-integration).
 * The team at @antmicro - specifically @kgugala might be able to help point you to the right set of resources.

PS Are you associated with the [BYU FPGA Bootcamp](https://byu-cpe.github.io/ComputingBootCamp/) or @taylor-bsg's group at University of Washington?

developandplay, 2021-06-23:
Thank you.

I watched your CCC talk on Symbiflow when I started getting into FPGAs and it's definitely an interesting project.
Currently I still have the ulx3s at home that I'd love to get Blackparrot to work on with Yosys. So it's definitely on my radar long term wise.

I think @dpetrisko talked to the surelog team a few weeks ago to see how compatible Blackparrot is at this point. So once that part looks good I think it's about time to get LiteX + BlackParrot + SymbiFlow going.

Also...Yep, I'm currently writing my Bachelor's thesis on Blackparrot+Litex. I'm glad @taylor-bsg agreed to supervise the project even though I study at an university on the other side of the world.

enjoy-digital, 2021-06-23:
Congrats @developandplay, that's great you managed to understand the issue and fix it. Let's say this was a good way to get into LiteX/CPU integration :) Future issues should be easier to understand and fix.

PR #953 (2021-06-23): Blackparrot 32bit csr
Fixes #938
closed 2021-06-23 by enjoy-digital