enjoy-digital/litex #939: Are --ram-init and --sdram-init meant to be drop in replacements?

developandplay, 2021-06-10:
Let's say I want to simulate my memory more accurately so I switch from `--ram-init` to `--sdram-init`.

Now apart from setting `--with-sdram`, `--sdram-module` and `sdram-data-width` is there anything I need to bear in mind?
I'm particularly thinking of whether the csr layout will change and whether I can use the same `boot.bin`.

developandplay, 2021-06-10:
Ok so as for the csr layout: it will change

enjoy-digital, 2021-06-14:
@developandplay: Both are very similar yes. One subble difference is that during DRAM initialization, a memtest is performed on hardware (so writing/reading data to the DRAM) and has to be disabled in simulation to avoid corrupting initial contents. This is the reason of this line: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_sim.py#L215

You can find a simulation with DRAM pre-initialized with Linux images and then booting to it in Linux-On-LiteX-VexRiscv repository: https://github.com/litex-hub/linux-on-litex-vexriscv/blob/master/sim.py#L106-L111

developandplay, 2021-06-14:
@enjoy-digital
I was playing around with Linux-On-LiteX-VexRiscv but the `--with-sdram` flag doesn't seem to do anything.
It seems like sdram is always enabled anyways.

If I try to make `--with-sdram` parameter work as in litex_sim then I'm getting some seamingly unrelated errors:
`AttributeError: 'VexRiscvSMP' object has no attribute 'cluster_name'`

developandplay, 2021-06-14:
Found the issue: I had to add a `VexRiscvSMP.generate_cluster_name()` because that's usually executed in `add_memory_buses`.

```
diff --git a/sim.py b/sim.py
index 1f8e889..3594385 100755
--- a/sim.py
+++ b/sim.py
@@ -97,7 +97,9 @@ class SoCLinux(SoCCore):
         sdram_module     = "MT48LC16M16",
         sdram_data_width = 32,
         sdram_verbosity  = 0,
-        with_ethernet    = False):
+        with_ethernet    = False,
+        with_sdram = False,
+        **kwargs):
         platform     = Platform()
         sys_clk_freq = int(100e6)
 
@@ -118,7 +120,8 @@ class SoCLinux(SoCCore):
             cpu_type                 = "vexriscv_smp",
             cpu_variant              = "linux",
             integrated_rom_size      = 0x8000,
-            uart_name                = "sim")
+            uart_name                = "sim",
+            **kwargs)
         self.add_config("DISABLE_DELAYS")
 
         # Add linker region for OpenSBI
@@ -130,26 +133,27 @@ class SoCLinux(SoCCore):
         self.add_csr("supervisor")
 
         # SDRAM ------------------------------------------------------------------------------------
-        sdram_clk_freq   = int(100e6) # FIXME: use 100MHz timings
-        sdram_module_cls = getattr(litedram_modules, sdram_module)
-        sdram_rate       = "1:{}".format(sdram_module_nphases[sdram_module_cls.memtype])
-        sdram_module     = sdram_module_cls(sdram_clk_freq, sdram_rate)
-        phy_settings     = get_sdram_phy_settings(
-            memtype    = sdram_module.memtype,
-            data_width = sdram_data_width,
-            clk_freq   = sdram_clk_freq)
-        self.submodules.sdrphy = SDRAMPHYModel(
-            module    = sdram_module,
-            settings  = phy_settings,
-            clk_freq  = sdram_clk_freq,
-            verbosity = sdram_verbosity,
-            init      = ram_init)
-        self.add_sdram("sdram",
-            phy           = self.sdrphy,
-            module        = sdram_module,
-            origin        = self.mem_map["main_ram"],
-            l2_cache_size = 0)
-        self.add_constant("SDRAM_TEST_DISABLE") # Skip SDRAM test to avoid corrupting pre-initialized contents.
+        if with_sdram:
+            sdram_clk_freq   = int(100e6) # FIXME: use 100MHz timings
+            sdram_module_cls = getattr(litedram_modules, sdram_module)
+            sdram_rate       = "1:{}".format(sdram_module_nphases[sdram_module_cls.memtype])
+            sdram_module     = sdram_module_cls(sdram_clk_freq, sdram_rate)
+            phy_settings     = get_sdram_phy_settings(
+                memtype    = sdram_module.memtype,
+                data_width = sdram_data_width,
+                clk_freq   = sdram_clk_freq)
+            self.submodules.sdrphy = SDRAMPHYModel(
+                module    = sdram_module,
+                settings  = phy_settings,
+                clk_freq  = sdram_clk_freq,
+                verbosity = sdram_verbosity,
+                init      = ram_init)
+            self.add_sdram("sdram",
+                phy           = self.sdrphy,
+                module        = sdram_module,
+                origin        = self.mem_map["main_ram"],
+                l2_cache_size = 0)
+            self.add_constant("SDRAM_TEST_DISABLE") # Skip SDRAM test to avoid corrupting pre-initialized contents.
 
         # Ethernet ---------------------------------------------------------------------------------
         if with_ethernet:
@@ -195,19 +199,31 @@ def main():
     parser.add_argument("--threads",              default=1,               help="Set number of threads (default=1)")
     VexRiscvSMP.args_fill(parser)
     args = parser.parse_args()
+    soc_kwargs = {}
 
     VexRiscvSMP.args_read(args)
+    VexRiscvSMP.generate_cluster_name()
     sim_config = SimConfig(default_clk="sys_clk")
     sim_config.add_module("serial2console", "serial")
     if args.with_ethernet:
         sim_config.add_module("ethernet", "eth", args={"interface": "tap0", "ip": args.remote_ip})
 
     for i in range(2):
+        if not args.with_sdram:
+            soc_kwargs["integrated_main_ram_size"] = 0x10000000 # 256 MB
+            soc_kwargs["integrated_main_ram_init"] = get_mem_data({
+                "images/Image":       "0x00000000",
+                "images/rv32.dtb":    "0x00ef0000",
+                "images/rootfs.cpio": "0x01000000",
+                "images/opensbi.bin": "0x00f00000"
+            }, "little")
         soc = SoCLinux( i!=0,
             sdram_module     = args.sdram_module,
             sdram_data_width = int(args.sdram_data_width),
             sdram_verbosity  = int(args.sdram_verbosity),
-            with_ethernet    = args.with_ethernet)
+            with_ethernet    = args.with_ethernet,
+            with_sdram       = args.with_sdram,
+            **soc_kwargs)
         if args.with_ethernet:
             for i in range(4):
                 soc.add_constant("LOCALIP{}".format(i+1), int(args.local_ip.split(".")[i]))
```

developandplay, 2021-06-14:
Now however the boot get's stuck on liftoff...

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS CRC passed (70432727)

 Migen git sha1: 3ffd64c
 LiteX git sha1: 444c4e01

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 100MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
MAIN-RAM:       262144KiB 

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Executing booted program at 0x40f00000

--============= Liftoff! ===============--
```

enjoy-digital, 2022-04-28:
--ram-init and --sdram-init have been reworked recently and Linux-on-LiteX-VexRiscv/LiteX-Sim also updated, so I think we can now close this issue since now obsolete.

closed 2022-04-28 by enjoy-digital