enjoy-digital/litex #946: Wrong time unit in trace files

navaneeth-cirel, 2021-06-16:
Hi, 

I am dumping the trace files in simulation using ``lxsim --cpu-type vexriscv --trace --sim-debug`` and its giving me the sim.vcd file but when I view the ```sys_clk``` its period is shown as 1ms. This does not change behavior of the simulation but it would be nice to see this as the same clock which is specified for sim which is 1MHz. 

Thanks
Navaneeth

navaneeth-cirel, 2021-06-17:
Adding screenshot

[image]

developandplay, 2021-07-16:
@navaneeth-cirel 
Does this occur for you with `lxsim --cpu-type vexriscv --trace` as well?
I tried `lxsim --cpu-type vexriscv --trace` and verilator is showing me 1us as expected.
Additionally which version of verilator are you using?

navaneeth-cirel, 2021-07-17:
@developandplay
Yes it is the same if I run `lxsim --cpu-type vexriscv --trace`  as well.
My verilator version is `Verilator 4.028 2020-02-06 rev v4.026-92-g890cecc1`. Can you tell me the version of Verilator which worked for you?

developandplay, 2021-07-17:
I'm using `Verilator 4.203 devel rev v4.202-43-gaba38830`.
So maybe this issue is already fixed in a newer versions of Verilator.

[image]
Just to make sure it's not an gtkwave issue: 
Which version of gtkwave are you using?
Could you try checking if [my trace file](https://github.com/developandplay/linux-on-litex-vexriscv/raw/time-unit-test/sim.fst) has the same issue for you?

navaneeth-cirel, 2021-08-03:
@developandplay 

I have checked your trace file and it is fine, so it is not a gtkwave issue.

I tried installing the latest verilator through conda https://anaconda.org/litex-hub/verilator , this installs `Verilator 4.211 devel rev UNKNOWN.REV` . With this the simulation is giving and error and not even starting. 
`x86_64-conda-linux-gnu-c++: error: unrecognized command line option '-fcf-protection=none'; did you mean '-flto-partition=none'?`

navaneeth-cirel, 2021-08-03:
@developandplay 

It does seem to be a Verilator issue, I installed Verilator from source and now the time unit is correct when I run `lxsim --cpu-type vexriscv --trace`. 
Somehow the Verilator installation through Conda has some issue with the simulation giving the above error.

developandplay, 2021-08-03:
@navaneeth-cirel 
Great that you could resolve the problem.
I personally don't use conda for package management so I'm not sure but I would assume your gcc version does not support the '-fcf-protection' flag.

navaneeth-cirel, 2021-08-03:
@developandplay 

Thank you for your support.

May not be gcc because in the same environment the Verilator was built from source and that is working fine for me.

I will mark this issue as closed.

Thanks,
Navaneeth

closed 2021-08-03 by navaneeth-cirel