enjoy-digital/litex #950: Make sure BIOS is recompiled correctly when switching CPU/Variant

enjoy-digital, 2021-06-20:
Switching between CPU types or CPU variants will not trigger a full recompilation and this has been causing weird issues. The workaround for now is to do a `rm -rf build` before the build but this should be handled automatically.

cr1901, 2021-06-20:
This could also be related to the proposed feature to store the actual CPU/SoC config in `identifier_mem`.

ewenmcneill, 2021-06-21:
An alternative to removing the build directory might be to make the build directory name dependent on things which might otherwise need a fully recompilation, eg dependent on the CPU type / variant.  litex-buildenv has done something like this for years:

https://github.com/timvideos/litex-buildenv/blob/f568cf34bff82deb8baf54d972ee1083dc51417a/Makefile#L144

which causes a switch of build directories if you change CPU / variant, but retains the work from previous builds of other types to make it faster to switch back and forth between CPUs / variants for testing.

Ewen

enjoy-digital, 2021-08-24:
The BIOS/libs will now automatically be fully rebuilt on CPU type/variant changes with https://github.com/enjoy-digital/litex/commit/44b223a9182318d58b9d4249d879f925dfd11e7d. Having the build name depends on the CPU type/variant would introduce some complexity which I don't think is necessary, so I just kept the same software directory name but added a detection on `variables.mak` changes to trigger a full rebuild.

closed 2021-08-24 by enjoy-digital