enjoy-digital/litex #954: RFC: devicetree support

hvegh, 2021-06-23:
Dear LiteX devs.

I enjoy LiteX very much, great work great concept! Hope I can help.

### Background:
Yesterday I naively experimented a bit with Litex - platformio - zephyr(rtos). The idea was simple: I would provide platformio with:
```
[env:development]
platform = https://github.com/hvegh/platform-litex.git
board = litex_vexriscv
framework = zephyr
```
and the firmware would be generated automatically. Zephyr would provide the goodness of networking stacks and I would concentrate on glueing the right components together using platformio.
While I am now able to compile and run simple "hello world" examples the current approach is not scalable.

### Main issue:
- Zephyr relies heavily on static DTS structures and configuration options which needed to be modified by hand in order to get runnable code. 

### Possible solutions:
- Extend the build system to generate DTS structures (relatively easy). Something like: the distributed Build system generates on the basis of DTSi.templates and CSR definitions DTSi definitions. Then all these DTSi definitions are combined in a DTS, and passed on to software generators. This should be relatively simple todo without alot of maintenance. Hopefully once a core or subsystem has defined a static DTSi.template:

So for example:
```
	ram0: memory@"%08x".memories.main_ram.base() {
		device_type = "memory";
		reg = < "0x%08x".memories.main_ram.base() "0x%08x".memories.main_ram.size() >;
	};
```

could become:
```
	ram0: memory@40000000 {
		device_type = "memory";
		reg = < 0x40000000 0x10000000 >;
	};
```

### Questions;
- I like the concept of DTS but it is maybe limited in adoption, do we like to see this idea explored in LiteX? (Main question)
- In the end I would like to leverage the biggest actively developed codebase on a LiteX softcore with the least amount of hassle.  Is Zephyr a good candidate for this? (Secondary question) It should be even possible to create driver stubs to work within a Arduino framework.

Regards,
Henk

mithro, 2021-06-23:
@hvegh - Take a look at the following;
 * https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_json2dts.py
 * https://github.com/litex-hub/litex-renode/blob/master/generate-zephyr-dts.py

@mateusz-holenko

mithro, 2021-06-23:
https://github.com/timvideos/litex-buildenv/wiki/DeviceTree might be interesting too.

enjoy-digital, 2021-06-24:
Hi @hvegh,

thanks for your feedback. Automatic DTS generation is indeed an approach that we are already experimenting and using on Linux-On-LiteX-VexRiscv project for example (and we currently extending it to the other Linux projectsintegrating different CPUs with LiteX: Rocket, Mor1kx, etc...). The current approach generates a `.json` file describing the LiteX SoC and then translate this `.json` file to a `.dts` file with `litex_json2dts.py` tool. This tool has only been used for now to generate `.dts` for Linux, but it would be really interesting to see if we can easily use it/adapt it to also generate `.dts` for Zephyr. As pointed by @mithro, this is only one approach and other approaches have also been experimented and directly tested with Zephyr.

If expected `.dts` for Linux and Zephyr are very similar, extending `litex_json2dts.py` and adding support for Zephyr to it would probably the more interesting approach. With the JSON export, it's also possible for any developer to create a custom tool that will be run after the build to generate specific translation of the JSON SoC description.

geertu, 2021-06-24:
@enjoy-digital 
> If expected `.dts` for Linux and Zephyr are very similar, extending `litex_json2dts.py` and adding support for Zephyr to it would probably the more interesting approach. With the JSON export, it's [...]

As DTS describes hardware, not software policy (except for some parameters in the /chosen node), there should be only a single DTS for the hardware, that can be used by whatever software is running on the hardware.

enjoy-digital, 2021-06-24:
Thanks @geertu, that's what I was guessing but was not sure there were possible small differences depending the OSes. That will simplify things for if there are no differences. Thanks for providing your/expertise on this.

mithro, 2021-06-24:
@geertu While I agree in theory about your DTS comment, in practice things are a bit lot more complicated. See for example the DTS patching things like uboot do and various incompatibilities between Linux *versions*.

mithro, 2021-06-24:
I *think* there is a potential future around LiteX generating DTS directly rather than via the JSON conversion method. 

Having DTS fragment generation "be close" to the hardware description seems like it would mean that the two are better kept in sync (IE when someone changes the hardware, they also update the required DTS fragment).

hvegh, 2021-07-02:
Sorry for my late reaction, thank you very much for the feedback. Currently I have again some playtime. I will see if I can enhance the workflow a little bit and glue the components together. 

I will experiment with the `litex_json2dts.py` as @geertu mentioned. 

I will keep you posted.

hvegh, 2021-07-05:
I have created a refactored and standalone version of the DTS overlay converter:
- https://github.com/litex-hub/litex-renode/blob/master/generate-zephyr-dts.py
  See this pull request:
  https://github.com/litex-hub/litex-renode/pull/34

I now understand a little bit better how DTS works. Using the generated DTS overlays it possible to generate working code examples on verilator and the fpga using platformio quite easy. 

**So the GOAL is: platformio + <board>_csr.json + src/code  => firmware.bin**

In order todo that I see:

- [ ]  Add platformio patch to allow for rv32im[a] compilation, see: https://github.com/platformio/zephyr/pull/9
- [ ]  Would be nice to add `generate-zephyr-dts.py` as part of https://github.com/zephyrproject-rtos/zephyr
- [ ]  Engage with the platformio community on how to deal with <board> specific variants.

There are some opportunities possible in order to make this available for a wider audience. 
   

 
Still some problems exist:
- 
 
 as easily for the verilator simulation as it is for the the other fpga boards 
process and it usage

mithro, 2021-07-07:
FYI - @mateusz-holenko