enjoy-digital/litex #963: Has Litex been used for a tapeout?

hadirkhan10, 2021-07-05:
Hi, I was going through the Litex repository. Does the litex project deals with building SoCs on the FPGAs only? Or has anyone taped-out an ASIC using Litex? Please mention any reference if anyone has done so.

enjoy-digital, 2021-07-05:
Hi @hadirkhan10,

The main focus of LiteX currently  is indeed to create designs/SoC on FPGAs, but the framework has already been used:
- To verify and stress RTL code that was going to be taped-out.
- To create ASIC hardware  benches to verify and stress the taped-out design.
- To create ASIC with very specific functionalities  (Interface bridges for example) for clients.

We are not able to share that much information about these project since are confidential (done by @enjoy-digital for clients) but this should give you first overview of what has been done with LiteX regarding ASICs.

In the long term, we are thinking about creating full ASiC SoCs with partners, as proof of concept first and maybe more seriously later but for now the focus is still on FPGAs.

Some groups seem to be using LiteX (or planning to) to create SoCs (ex LibreSoC / ChipEleven) but we aren't directly involved in these projects.

hadirkhan10, 2021-07-05:
Thanks! that pretty much answers what I was looking for :)

closed 2021-07-05 by hadirkhan10