enjoy-digital/litex #968: Replacing libbase with newlib or picolibc?

alanvgreen, 2021-07-12:
Could libbase be replaced with [newlib](https://sourceware.org/newlib/) or [picolibc](https://github.com/picolibc/picolibc)? Using one of the more common libraries would simplify porting software to LiteX SoCs, and sidestep issues such as #322 

In [a CFU-Playground issue](https://github.com/google/CFU-Playground/issues/142), @mithro wrote "Modern compilers + LTO removes most of the reason for libbase to even exist."

Is a newlib or picolibc worth pursuing? How complex would it be?

enjoy-digital, 2021-07-13:
Hello @alanvgreen,

with modern compilers, we could indeed think of it and try to revisit/simplify LiteX BiOS. We could even think of using something like `zig cc` provided as a python package  that would provide all we need (libc and compiler_rt, but would need to be extended to risv32, lm32 and or1k): 
- https://andrewkelley.me/post/zig-cc-powerful-drop-in-replacement-gcc-clang.html 
- https://pypi.org/project/ziglang/

I see it more as a long term task and would be happy to test/review libbase remplacements.

mithro, 2021-07-14:
FYI - @mateusz-holenko

enjoy-digital, 2021-09-27:
#976 has been merged (see https://github.com/enjoy-digital/litex/pull/976#pullrequestreview-764322328). I did some changes to minimize changes for users:
- https://github.com/enjoy-digital/litex/commit/ae1d43b9658d9717abbd35fa3754b979310e3369
- https://github.com/enjoy-digital/litex/commit/3d32ac3d2e5f0774513367770e5f78557f286d21

picolibc dependency can be installed with: `litex_setup.py init update install` and Meson is now required: `pip3 install meson`.

closed 2021-09-27 by enjoy-digital