enjoy-digital/litex #972: Add network interface information to LiteScope wiki

developandplay, 2021-07-19:
When using LiteScope it is necessary that the network interface of the host computer responds to requests to 192.168.1.100 as that's the default setting.

Therefore adding a line to the [wiki article](https://github.com/enjoy-digital/litex/wiki/Use-LiteScope-To-Debug-A-SoC) like:

> Make sure your host computer responds to requests at 192.168.1.100 by running:
> `sudo ip addr add 192.168.1.100/24 scope global dev <interface>`

would be a great help.

enjoy-digital, 2021-07-20:
Hi @developandplay,

when using Etherbone, the IP address of the Host just need to be on the same network but does not necessarily needs to be `192.168.1.100`. Can you describe the issue you had?

developandplay, 2021-07-20:
I'm consistently getting a timed out error if I don't add it.
This might be also an issue with my network setup but that's what fixes it.
I started using this command for TFTP netboot but for litescope it seems to be helping too.
The command is from: https://github.com/litex-hub/linux-on-litex-rocket/blob/master/README.md

```
$ litex_server --udp --udp-ip=192.168.1.50
[CommUDP] ip: 192.168.1.50 / port: 1234 / tcp port: 1234
Traceback (most recent call last):
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 48, in probe
    datas, dummy = self.socket.recvfrom(8192)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/martin/.local/bin/litex_server", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'litex_server')()
  File "/home/martin/coding/litex/litex/litex/tools/litex_server.py", line 270, in main
    server.open()
  File "/home/martin/coding/litex/litex/litex/tools/litex_server.py", line 90, in open
    self.comm.open()
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 31, in open
    self.probe(self.server, self.port)
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 56, in probe
    raise Exception(f"Unable to probe Etherbone server at {self.server}.")
Exception: Unable to probe Etherbone server at 192.168.1.50.
$ sudo ip addr add 192.168.1.100/24 scope global dev wlp3s0
$ litex_server --udp --udp-ip=192.168.1.50
[CommUDP] ip: 192.168.1.50 / port: 1234 / tcp port: 1234
$ sudo ip addr del 192.168.1.100/24 scope global dev wlp3s0
$ litex_server --udp --udp-ip=192.168.1.50
[CommUDP] ip: 192.168.1.50 / port: 1234 / tcp port: 1234
Traceback (most recent call last):
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 48, in probe
    datas, dummy = self.socket.recvfrom(8192)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/martin/.local/bin/litex_server", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'litex_server')()
  File "/home/martin/coding/litex/litex/litex/tools/litex_server.py", line 270, in main
    server.open()
  File "/home/martin/coding/litex/litex/litex/tools/litex_server.py", line 90, in open
    self.comm.open()
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 31, in open
    self.probe(self.server, self.port)
  File "/home/martin/coding/litex/litex/litex/tools/remote/comm_udp.py", line 56, in probe
    raise Exception(f"Unable to probe Etherbone server at {self.server}.")
Exception: Unable to probe Etherbone server at 192.168.1.50.
$ sudo ip addr add 192.168.1.100/24 scope global dev wlp3s0
$ litex_server --udp --udp-ip=192.168.1.50
[CommUDP] ip: 192.168.1.50 / port: 1234 / tcp port: 1234
Connected with 127.0.0.1:39114
```

mithro, 2021-07-20:
BTW You might want to check tcpdump to see the packets going past. wireshark is a great way to do this.