enjoy-digital/litex #973: How to deal with region overlap

developandplay, 2021-07-19:
When trying to add ethernet to BlackParrot on my Arty board I'm running into an region overlap issue:

```
ERROR:SoCBusHandler:Region overlap between ethmac and csr:
ERROR:SoCBusHandler:Origin: 0x50000000, Size: 0x00002000, Mode: RW, Cached: False Linker: False
ERROR:SoCBusHandler:Origin: 0x50000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False
```

How can I layout the regions such that I don't run into this issue?

enjoy-digital, 2021-07-20:
Hi @developandplay,

I just applied the same changes I did to Microwatt for a similar issue: https://github.com/enjoy-digital/litex/commit/c64c6e40564daee3014a21659926a3a2b922a6ad

I will try to find a better solution in the long term, but wil allow you to build your design.

developandplay, 2021-07-20:
Thanks for the quick fix @enjoy-digital !
I'll try `netboot` soon and close the issue.

developandplay, 2021-07-26:
Works!
There is some other issue now when using litescope but I can make it work with this knowledge.

closed 2021-07-26 by developandplay