enjoy-digital/litex #984: Handling of mem_map when cpu type is "None"

tcal-x, 2021-07-30:
In the Fomu workshop, one of the exercises uses LiteX to create a WishBone bus with peripherals (RAM and RGB LED) but _no_ CPU.   This is probably a rare situation that is not much tested.   It it part of the reason why we haven't been able to bump litex_boards submodule in the Fomu workshop (https://github.com/im-tomu/fomu-workshop/issues/411).

But now I'm seeing a related issue even with the current submodules.   CSRs are supposed to be mapped at 0xE0000000 / 0x60000000 ([link](https://github.com/litex-hub/litex-boards/blob/929e55d7e62b7430b917d8a28604f315d4bb64b5/litex_boards/targets/fomu.py#L155)), but after the build, they appear at 0x00000000 (see https://github.com/im-tomu/fomu-workshop/pull/509).    0x00000000 *should* be the ROM region.

I think this behavior originates where the CPU mem_map is [merged](https://github.com/enjoy-digital/litex/blob/1ce48a973b13ab6e0434408265133bcfc28fbdbb/litex/soc/integration/soc.py#L899) with the SOC's mem_map:
```
        self.mem_map.update(self.cpu.mem_map) # FIXME
```

If I print out the SOC mem_map, the CPU mem_map, and then the SOC mem_map after the update, I see:
```
soc memmap
{'rom': 0, 'sram': 268435456, 'spiflash': 536870912, 'main_ram': 1073741824, 'csr': 3758096384}
cpu memmap
{'csr': 0}
new soc memmap
{'rom': 0, 'sram': 268435456, 'spiflash': 536870912, 'main_ram': 1073741824, 'csr': 0}
```

If I add a guard as below, it prevents the CSR region base from being set to zero, and normal behavior is restored.
```
        if not isinstance(self.cpu, cpu.CPUNone):
            self.mem_map.update(self.cpu.mem_map) # FIXME
```

Of course, there may be a better fix (such has having an empty mem_map for the CPU when its type is None).

enjoy-digital, 2021-07-30:
Thanks for the feedback @tcal-x, I indeed wanted to rework this and at least add the override info to the build log (=full traduction of the FIXME I put some time ago :)).

I just used the Fomu target and should have fixed your issue with https://github.com/enjoy-digital/litex/commit/ce5864983d9432fa8a3ce57516a89ee4156be0e5.

The CPUNone case is a bit different than others, so I changed the priority in the remapping for it: User's mapping have the priority in this case.

Please let me know if it fixes the issue on your side (and if you have other blockers preventing you from using upstream LiteX).

enjoy-digital, 2021-08-25:
As described in my previous comment, it should be fixed. I'm closing this but feel free to re-open @tcal-x if it's not the case.

closed 2021-08-25 by enjoy-digital