enjoy-digital/litex #985: Common UART for UARTBone and shell interface

navaneeth-cirel, 2021-08-03:
Hi,

Is it possible to have a common PHY for a UART connecting to CPU (Shell/Terminal interface) and a UART to Wishbone bridge. Is this what the `UARTCrossover` aims to provide? I am a little confused about its usage. 
Any help is appreciated.

Thanks
Navaneeth

cklarhorst, 2021-08-11:
Hi,
- `uartbone` or `bridge` is the UART/Wishbone bridge
- `crossover` is a UART implementation that can be accessed through Wishbone 
- `crossover+bridge` generates the uartbone bridge on the rx/tx pins and also adds the crossover implementation to the SoC (which is then used by the CPU)

So I think `crossover+bridge` is what you are looking for. The `tools/litex_server` script can be used to access the uartbone, `tools/litex_term` provides access to the crossover UART.

https://github.com/enjoy-digital/litex/wiki/Use-Host-Bridge-to-control-debug-a-SoC might also help you.

navaneeth-cirel, 2021-08-12:
Hi @cklarhorst ,

Thank you for your inputs, I will try the `crossover+bridge`.

enjoy-digital, 2022-05-27:
Closing since answered.

closed 2022-05-27 by enjoy-digital