enjoy-digital/litex #989: femtorv fails to initialize memory

michalsieron, 2021-08-11:
While I was working on #976 and #988 I noticed that femtorv has some issues with memory when booted on board.

Seems like it isn't my fault, as it also happens on master branch.

Steps:
1. `python3 litex-boards/litex_boards/targets/digilent_arty.py --csr-json build/csr.json --cpu-type femtorv --build`
2. `python3 litex-boards/litex_boards/targets/digilent_arty.py --csr-json build/csr.json --cpu-type femtorv --load`

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2021 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Aug 10 2021 13:51:07
 BIOS CRC passed (9175dab1)

 Migen git sha1: 3ffd64c
 LiteX git sha1: 79ac0931

--=============== SoC ==================--
CPU:        FemtoRV @ 100MHz
BUS:        WISHBONE 32-bit @ 4GiB
CSR:        32-bit data
ROM:        128KiB
SRAM:       8KiB
L2:     8KiB
SDRAM:      262144KiB 16-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000001111111111111110000| delays: 20+-07
  m0, b02: |00000000000000000000000000000111| delays: 30+-01
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 20+-07
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000001111111111111110000| delays: 20+-07
  m1, b02: |00000000000000000000000000000111| delays: 30+-01
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: 20+-07
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 7735/8192
  data errors: 45776/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex>
```

`lxsim` doesn't check memory so no issue there
No problems when booted with `renode`

enjoy-digital, 2021-08-11:
@michalsieron: I indeed also saw this while testing on hardware when doing the initial FemtoRV support and haven't yet investigated it. I'm not sure if it's an issue on the memory interface or on the core itself. To investigate, it could be interesting to enable memory test with `lxsim` (that is disabled to speed up simulation).

mithro, 2021-08-11:
FYI - @umarcor @BrunoLevy @JosephBushagour

enjoy-digital, 2021-11-19:
This is fixed with https://github.com/enjoy-digital/litex/commit/54a4e6c1f626ffe86aabfb46c447b1afe1641440. Contrary to traditional buses, FemtoRV is not maintaining values on the bus after the first cycle when waiting for access completion.

closed 2021-11-19 by enjoy-digital

BrunoLevy, 2021-11-19:
Fantastic ! (wow, this did not take long !!).  

The other variants are:
- tachyon (RV32I like quark, but one additional cycle in state machine, supports higher maxfreq, not very interesting)
- electron (RV32IM)
- intermissum (RV32IM + interrupts + CSR)
- gracilis (RV32IMC + interrupts + CSR)
- petitbateau (RV32IMFC + interrupts + CSR)