enjoy-digital/litex #990: Ibex synthesis fails

michalsieron, 2021-08-11:
I noticed that Ibex synthesis fails with following errors:

```
ERROR: [Synth 8-1766] cannot open include file dv_fcov_macros.svh [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:11]
ERROR: [Synth 8-2715] syntax error near ( [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:850]
ERROR: [Synth 8-2715] syntax error near ( [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:850]
ERROR: [Synth 8-2715] syntax error near ( [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:852]
WARNING: [Synth 8-2841] use of undefined macro DV_FCOV_SIGNAL [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:850]
ERROR: [Synth 8-2715] syntax error near ( [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:854]
ERROR: [Synth 8-2715] syntax error near ( [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:855]
ERROR: [Synth 8-2715] syntax error near & [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:856]
WARNING: [Synth 8-2611] redeclaration of ansi port debug_req_i is not allowed [/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv:856]
Failed to read verilog '/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv'
```

I tried changing some include paths
```diff
diff --git a/pythondata_misc_opentitan/resources/hw/ip/aes/dv/cov/aes_cov_if.sv b/pythondata_misc_opentitan/resources/hw/ip/aes/dv/cov/aes_cov_if.sv
index 1d504ec46..afb4aac15 100644
--- a/pythondata_misc_opentitan/resources/hw/ip/aes/dv/cov/aes_cov_if.sv
+++ b/pythondata_misc_opentitan/resources/hw/ip/aes/dv/cov/aes_cov_if.sv
@@ -12,7 +12,7 @@ interface aes_cov_if
   import uvm_pkg::*;
   import aes_pkg::*;
   import dv_utils_pkg::*;
-  `include "dv_fcov_macros.svh"
+  `include "../../../../dv/sv/dv_utils/dv_fcov_macros.svh"

   bit          en_full_cov = 1'b1;
   bit          en_intg_cov = 1'b1;
diff --git a/pythondata_misc_opentitan/resources/hw/ip/pattgen/dv/cov/pattgen_cov_if.sv b/pythondata_misc_opentitan/resources/hw/ip/pattgen/dv/cov/pattgen_cov_if.sv
index a1bf1c63d..026fe117b 100644
--- a/pythondata_misc_opentitan/resources/hw/ip/pattgen/dv/cov/pattgen_cov_if.sv
+++ b/pythondata_misc_opentitan/resources/hw/ip/pattgen/dv/cov/pattgen_cov_if.sv
@@ -9,7 +9,7 @@ interface pattgen_cov_if (

   import uvm_pkg::*;
   import dv_utils_pkg::*;
-  `include "dv_fcov_macros.svh"
+  `include "../../../../dv/sv/dv_utils/dv_fcov_macros.svh"

   bit en_full_cov = 1'b1;
   bit en_intg_cov = 1'b1;
diff --git a/pythondata_misc_opentitan/resources/hw/ip/uart/dv/cov/uart_cov_if.sv b/pythondata_misc_opentitan/resources/hw/ip/uart/dv/cov/uart_cov_if.sv
index 0e74f0414..0cd589779 100644
--- a/pythondata_misc_opentitan/resources/hw/ip/uart/dv/cov/uart_cov_if.sv
+++ b/pythondata_misc_opentitan/resources/hw/ip/uart/dv/cov/uart_cov_if.sv
@@ -9,7 +9,7 @@ interface uart_cov_if (

   import uvm_pkg::*;
   import dv_utils_pkg::*;
-  `include "dv_fcov_macros.svh"
+  `include "../../../../dv/sv/dv_utils/dv_fcov_macros.svh"

   bit en_full_cov = 1'b1;
   bit en_intg_cov = 1'b1;
diff --git a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv
index 1a6523c40..3d5485484 100644
--- a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv
+++ b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_controller.sv
@@ -8,7 +8,7 @@
  */

 `include "prim_assert.sv"
-`include "dv_fcov_macros.svh"
+`include "../../../dv/sv/dv_utils/dv_fcov_macros.svh"

 module ibex_controller #(
     parameter bit WritebackStage  = 0,
diff --git a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_id_stage.sv b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_id_stage.sv
index 7107ee65f..e0d87b906 100644
--- a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_id_stage.sv
+++ b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_id_stage.sv
@@ -15,7 +15,7 @@
  */

 `include "prim_assert.sv"
-`include "dv_fcov_macros.svh"
+`include "../../../dv/sv/dv_utils/dv_fcov_macros.svh"

 module ibex_id_stage #(
     parameter bit               RV32E           = 0,
diff --git a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_load_store_unit.sv b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_load_store_unit.sv
index a26a3cbdb..b317601ff 100644
--- a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_load_store_unit.sv
+++ b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_load_store_unit.sv
@@ -12,7 +12,7 @@
  */

 `include "prim_assert.sv"
-`include "dv_fcov_macros.svh"
+`include "../../../dv/sv/dv_utils/dv_fcov_macros.svh"

 module ibex_load_store_unit
 (
diff --git a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_wb_stage.sv b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_wb_stage.sv
index 02f1308e1..c982ebd2b 100644
--- a/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_wb_stage.sv
+++ b/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_wb_stage.sv
@@ -12,7 +12,7 @@
  */

 `include "prim_assert.sv"
-`include "dv_fcov_macros.svh"
+`include "../../../dv/sv/dv_utils/dv_fcov_macros.svh"

 module ibex_wb_stage #(
   parameter bit WritebackStage = 1'b0
```

but then I got different set of warnings and errors and I gave up
```
ERROR: [Synth 8-3438] module 'ibex_core' declared at '/home/michal/dev/litex/pythondata-misc-opentitan/pythondata_misc_opentitan/resources/hw/vendor/lowrisc_ibex/rtl/ibex_core.sv:15' does not have any parameter 'RegFile' used as named parameter override [/home/michal/dev/litex/build/digilent_arty/gateware/digilent_arty.v:12968]
ERROR: [Synth 8-6156] failed synthesizing module 'digilent_arty' [/home/michal/dev/litex/build/digilent_arty/gateware/digilent_arty.v:4]
```

enjoy-digital, 2021-08-13:
Thanks for the feedback @michalsieron, this is probably related to changes in Ibex that has evolved since the initial support in LiteX. 

Can you try to switch opentitan in to the version used at the time of the port? (https://github.com/enjoy-digital/litex/commit/510bda4c9939f9f025c278e2d04214fe8d6769b6) 

If working, can you do a PR to specify the version of opentitan that has to be used in `litex_setup.py`? (https://github.com/enjoy-digital/litex/blob/master/litex_setup.py#L47)

michalsieron, 2021-08-13:
I guess that commit would be this one? https://github.com/litex-hub/pythondata-misc-opentitan/commit/e43566c9e68040e51fa652df8cca5a4f91b609e7

And indeed, it does build.
In fact this is the latest working commit without issue present on current master: https://github.com/litex-hub/pythondata-misc-opentitan/commit/e0af01e9a68e4b81c5aa34b616121619696bec6b

However neither of them works properly when loaded.
It seems that some things work, as LEDs on my Arty A7 start blinking, but nothing gets printed.
I checked with LiteX on master as well as some older commits, but either results are the same or it gets too old and then litex-boards fail, because support for Arty A7 was added later.

enjoy-digital, 2021-09-08:
@michalsieron: I just tested Ibex on Arty with: `./digilent_arty.py --cpu-type=ibex --sys-clk-freq=50e6 --integrated-main-ram=0x1000 --build --load`
And it works fine with the following bitstream: [digilent_arty_ibex_2021_09_08.zip](https://github.com/enjoy-digital/litex/files/7130030/digilent_arty_ibex_2021_09_08.zip)

We have to reduce sys-clk-freq with Ibex to meet timing, this was maybe the issue when you tested. So I'm closing, but feel free to re-open if you still have the issue with the command I shared.

closed 2021-09-08 by enjoy-digital

michalsieron, 2021-09-08:
Yeah, it works now, though memory init fails.
But from my previous tests, it fails on other CPUs as well after lowering clock speed. So that is probably related to https://github.com/litex-hub/litex-boards/issues/166 and #933

enjoy-digital, 2021-09-08:
@michalsieron: Indeed, I'm pretty sure it's the same issue. I'll try to look at it soon.