enjoy-digital/litex #996: CPUs keep resetting on PYNQ-Z2

developandplay, 2021-08-11:
I'm adding the PYNQ-Z2 to LiteX.
The process was pretty straight forward as I could base it on the Zybo Z7.
Unfortunately any CPU I use gets reset every 2-3 seconds.
By that I mean the ledchaser will get reset and UART as well just like when pressing the reset button.
I tried setting `pll.reset.eq(0)` but it does not seem to be affecting the RTL.

enjoy-digital, 2021-08-13:
Hi @developandplay,

I assume you are creating a SoC with a Soft Core CPU and not the PS. If so, this is very similar to a regular FPGA and the SoC just need a stable clock to operate correctly. You could first disconnect the reset. If still not working, I would recommend checking the clock and be sure that it's stable during the reset.

While answering, I'm just looking at Pynq Z2 schematic and it seems the clock from the PL is coming from the Ethernet PHY. If so, the issue could be similar to an issue we had on the Pano Logic G2: the LiteX SoC while initializing the Ethernet PHY was reseting it and then stopping the main clock source. To workaround this, you can comment out `rst_n `pad of the Ethernet PHY and drive it manually as we are doing on the Pano Logic G2: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/pano_logic_g2.py#L36

developandplay, 2021-08-13:
@enjoy-digital Thanks for the second hint.
Just read the PYNQ Z2 manual again and it seem like the resetting Ethernet PHY is indeed the problem.
Apparently if the ethernet is connected then the board won't reset the PHY.
https://forums.xilinx.com/t5/Other-FPGA-Architecture/MMCM-Losing-Lock-on-a-PYNQ-Z2-Zynq-7020/td-p/923525

closed 2021-08-13 by developandplay

enjoy-digital, 2021-08-13:
@developandplay: Great!