enjoy-digital/litex #997: Issue in configuring Main Ram Size.

asadaleem-rs, 2021-08-12:
I have generated an SoC from LiteX comprising of PicoRV32 as a CPU with all other parameters set to default. I need to configure the main ram size for which I used the parameter "**--integrated-main-ram-size**" However, this parameter doesn't seem to work at my end since it neither decreases nor increases the size of the main ram as shown once an SoC is generated. Could anyone please help me figure out if the RAM size is configurable or not. If it is configurable which parameter shall I use to alter its size?

michalsieron, 2021-08-12:
It seems that RAM usage reported at the and uses `sram`

https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/memusage.py#L25

So to change it you would use `--integrated-sram-size` option

asadaleem-rs, 2021-08-12:
As far as SRAM is concerned, I was able to configure its size as I have attached the image showing the command to configure SRAM and resulting in the SRAM generated of the specified size. However, I am talking specifically about the main ram size. (I am trying to synthesize the SoC generated using Vivado 2017.4 and it consumes a lot of resources.
[image]
)

michalsieron, 2021-08-12:
When you simulate it with `lxsim`, `main-ram` size seems to be predefined.
https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_sim.py#L328

asadaleem-rs, 2021-08-13:
Yes you are right, its size is predefined yet I am looking for a way to configure it.

enjoy-digital, 2021-08-13:
@asadaleem-rs : With `lxsim`, it's `main-ram` size is indeed currently forced to 256MB when no SDRAM is used. We could adapt the code to allow `main-ram-size` configuration but this is not currently supported, for now you can modify the line pointed by @michalsieron to reduce it.

Also, the generated SoC is intended for simulation, so running synthesis on it could indeed produce non-optimal/non-functional synthesis. For a proper synthesis, you have to create a platform/target for your board or look at the LiteDRAM/LiteEth/LitePCIe generators.

asadaleem-rs, 2021-08-13:
Thanks for the clarification.

asadaleem-rs, 2021-08-30:
I tried to resolve this issue by enabling the "main ram" to get arguments from the command line and it is working to the extent I have tested it.

closed 2021-08-30 by asadaleem-rs