enjoy-digital/litex #1001: Endianness in WishboneDMAReader

navaneeth-cirel, 2021-08-13:
Hi,

Shouldn't this be opposite ? 
https://github.com/enjoy-digital/litex/blob/78c1751c4781ffe156128748810ee4af85fee058/litex/soc/cores/dma.py#L19-L20

Just to try this out I used the `WishboneDMAReader` in a module of my SoC (with vexriscv) simulation to read a word from SRAM and print it in my C code and it printed reversed. The complete system is little endian.
```
DMA Read from : 0x10003fac
Readback from DMA: 0xdf0feca
Actual value : 0xcafef00d
```

with the format_bytes reversed
```python
 def format_bytes(s, endianness): 
     return {"little": s, "big": reverse_bytes(s)}[endianness]
```
the print is as expected

```
DMA Read from : 0x10003fac
Readback from DMA: 0xcafef00d
Actual value : 0xcafef00d
```

enjoy-digital, 2021-08-14:
Hi @navaneeth-cirel,

this probably similar to https://github.com/enjoy-digital/litedram/issues/251. There is a endianness mismatch between the model and real hardware. I'll try to address this soon.

navaneeth-cirel, 2021-08-14:
Hi @enjoy-digital ,

After brief look at the issue referenced it seems that there is a mismatch between the simulation and target hardware, but in my case I am seeing the mismatch in Simulation itself.
Just to give more clarity , what I am doing is writing a 32bit value to a `uint32_t` variable in the C code and set its address to the DMA to read and transfer to a CSR status register which I am then reading and printing in the C code, where the mismatch is observed.