enjoy-digital/litex #1005: Flashing/program via PCIe

dannycoin, 2021-08-17:
Hi there..
1) can an Acorn CLE 215+ be flashed through PCIe without JTAG cable? 
2) is it mandatory to use a JTAG cable to flash/program? 
3) Is there a guide to flashing via PCIe?

enjoy-digital, 2021-08-24:
Hi @dannycoin,

It's possible to update the SPI Flash over PCIe (and then update the bistream) with the LiteX design yes.

When building the Acorn CLE 215+ with PCIe (`python3 -m litex_boards.targets.sqrl_acorn --with-pcie --build --driver --flash`), the SPI Flash bridge will be instantiated:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sqrl_acorn.py#L120-L124

With the LitePCIe driver/utilities, you can then update the bistream with:
```
./litepcie_util flash_write your_bitstream.bin 0
./litepcie_util flash_reload
```

And then unmount the litepcie driver, do a PCIe rescan and mount the driver or reboot your machine.

So with this, you only need JTAG access to flash the first FPGA bitstream and can then just update over PCIe. This is used on several LitePCIe designs deployed on clients's site where programming over JTAG is not convenient.

enjoy-digital, 2021-09-08:
Closing since has been answered. This has been labelled as add-answer-to-wiki and answer will be added to Wiki.

closed 2021-09-08 by enjoy-digital