enjoy-digital/litex #1025: Migrate from "bus master" and "bus slave" terminology

alanvgreen, 2021-09-04:
In the computing context, many find the terms "master" and "slave" problematic and there have been many parallel efforts to migrate away from them. Here's (a list](https://en.wikipedia.org/wiki/Master%2Fslave_(technology)#Possible_replacement_conventions)..

In AMBA AXI documentation, "manager" is replacing "master" and "subordinate" is replacing "slave" ([reference](https://developer.arm.com/documentation/102202/latest/)). 

Would you be open to PRs updating the LiteX API to use the terms "manager" and "subordinate"?

enjoy-digital, 2021-09-06:
Thanks @alanvgreen, I've indeed been thinking about it but haven't been able to spend the time to really think about the best replacements, the way to ensure retro-compatibility methods call/imports. (And the issue that by providing retro-compatibility, part of the problem is not fully addressed since these terms will still be in the code base... but at least not the default API).

So it's a bit more than replacing the terms; for these replacements to have a real meaning we also need to update all shared projects/examples code. So that's something I have a preference to address myself since this is more than just doing the replacement in the code base. I'll think a bit more about it and do my best to plan some time in the future to do it. Until this, I just want to make it clear that these terms have always been used in the project as technical terms (and not being native English speaker they always had a technical resonance in my mind).

With this project and the support I'm providing, I'm also trying to encourage developers to ask questions, provide feedback, contribute in a friendly and respectful manner since I think every-one can contribute and can improve the project while also enjoying working on it. And while doing so, I've been curious to see some open-source projects, companies, reacting promptly to these social events/changes while not necessarily being exemplar on other aspects that are important for in me in LiteX... 

So these topics are difficult ones if you want to do things correctly, and I'd like LiteX to follow my intuitions on this. I of course understand the concerns, will try to address this in a not too far future, but need to plan it and some time for it. I hope the community can understand this.

Regards,

Florent

alanvgreen, 2021-09-10:
@enjoy-digital Thank you for your thoughtful response. 

Agreed that retro-compatibility is an important consideration. Yes, I imagine that the new and deprecated names would coexist for some time - apologies that I didn't make that clear.

When you have had time to decide a plan, please let me know - I'd be happy to help.

alanvgreen, 2021-09-14:
@enjoy-digital I've been thinking more about your response, and realized that I didn't acknowledge some of the points you made.

First, I 100% agree these changes are difficult. These terms are used to describe abstract concepts that have no close analogue in the everyday physical world, and they have permeated the entire industry. Changing everything will take time and effort and may never be complete.

I think the industry should prepare for two iterations on this problem. First, a change of terms to address the immediate problem, and avoid unnecessary offence and confrontation toward underrepresented groups. Second, when the industry agrees on new standard terms, an iteration where we move to those new terms. Both these iterations will be messy and difficult, because industry-wide API changes always are.

It's great that ARM has set a direction here. If we're lucky that will help avoid a second iteration.

Second, I think there is value and meaning in delivering an opt-in API-only change as a first step. For example, it LiteX provided alternatives to `bus.add_master()` and `bus.add_slave`, that would allow one of my projects to completely avoid the use of these terms. 

I'm also a fan of deeper and more meaningful changes to LiteX and, under your direction, I'd be happy to help.

In terms of timeframe if we were able to completely remove "slave" and "master" from our project's codebase in the next six months, that would be great. If it took three years, I'd be sad. Most of the points in between are somewhat acceptable :)

Finally, I wanted to be explicit: I love the helpful and respectful community that you have put together around LiteX - it's one of LiteX's best features. I very much appreciate the time and effort you on encouraging and modeling good behavior. Thank you.

enjoy-digital, 2021-09-15:
Thanks @alanvgreen, I also think there are indeed multiple steps to fully address this:
1) Provide a new recommended API that will avoid using the terms when using the project (but still allow building designs with the previous API to avoid breaking all existing designs).
2) Start updating internal LiteX code to avoid these terms and make the new API the default (and update all open-source examples/designs).
3) Add a recommendation warning to switch to the new API (but still allow the old one) and in this warning, provide the release that will deprecate it so that users start updating their designs.
4) Deprecate the old-API.

I think we can do 1) soon. Before doing this I just need to identify the different API/Modules for which we'll have the provide a new API. Sadly the terms are used in lots of places in Digital designs/LiteX: Bus definitions/components, Stream directions, SPI/I2C modules, etc... I'll try to start looking at this soon.  Doing 1) and 2) in the next few weeks/months will be possible/reasonable. We could then do 3) in after the last 2021 release and  fully deprecate the old API maybe 1 year later. I hope this gives you visibility on timeline I have in mind for this.

enjoy-digital, 2021-09-28:
While doing theses changes, we should also rename the default git default branch that is also problematic.

bunnie, 2021-09-28:
> While doing theses changes, we should also rename the default git default branch that is also problematic.

This is a minor point, but partially changing the default branch is slightly problematic for the CI builds we do. We do a recursive pull from `master` right now on all Litex dependencies, but Antmicro named their picolibc default `main`, which requires us to special-case the pull update to prevent the CI script from failing.

If you decide to only partially update the default branches (e.g. just change it for `litex` but leave the dependencies on master, and/or only update some of the dependencies), it would be helpful if you can publish a summary tracker of which brarches you've changed, and which braches you haven't, so we can special-case our default branch mappings accordingly, instead of guessing and failing or even worse getting stuck on a stale default without us being aware of the issue.

enjoy-digital, 2021-09-28:
@bunnie: In fact this morning I renamed the branch on the `picolibc` repo to `master` to avoid mixing  `master` and `main` for now. Once we'll do the switch, we'll do it on all repositories to avoid mixing terms and will indeed share this information with users. Thanks for the feedback.

alanvgreen, 2022-09-17:
Just checking in: any progress on the recommended API? If it helps, I can spend some time identifying API/Modules.

enjoy-digital, 2022-09-18:
@alanvgreen: Sorry, I haven't yet been able to spent time on this. If you have time to first identify the additionnal methods/classes we could add to avoid the direct use of these terms in your code, we could try to add them as a first step as discussed.

alanvgreen, 2022-10-31:
Hi @enjoy-digital ! 

We've been able to pare the problem areas back to these four. 

- [ ] `litex.soc.interconnect.wishbone.Arbiter(... masters= ...)`
    - Provide `controllers` alias for `masters` parameter
- [ ] `SoCBusHandler.add_master(master=...)`
    - Provide alias for method `add_controller(controller=...)`
- [ ] `SoCBusHandler.add_slave(slave=...)`
    - Provide alias for method `add_peripheral(peripheral=...)`
- [ ] `litespi.LiteSPI(... with_master= ...)`
    - Provide `with_controller` alias for `with_master` parameter

If these proposed aliases look acceptable to you, I'm  happy to send PRs. Also happy to moderately extend the scope of these changes if that would make sense.

enjoy-digital, 2022-10-31:
Hi @alanvgreen,

this looks good yes, you can send a PR, thanks.