enjoy-digital/litex #1027: Installation Error on Ubuntu 20.04 with Python 3.8

wuhanstudio, 2021-09-07:
Hi,

I followed instructions here https://github.com/enjoy-digital/litex/wiki/Installation:

```
$ wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
$ chmod +x litex_setup.py
$ ./litex_setup.py init install --user
$ ./litex_setup.py gcc
$ sudo apt install libevent-dev libjson-c-dev verilator
$ lxsim --cpu-type=vexriscv
```

Then it gives the error:

```
Traceback (most recent call last):
  File "/home/wuhanstudio/.local/bin/lxsim", line 11, in <module>
    load_entry_point('litex', 'console_scripts', 'lxsim')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/wuhanstudio/litex/litex/tools/litex_sim.py", line 17, in <module>
    from litex.build.sim import SimPlatform
  File "/home/wuhanstudio/litex/litex/build/sim/__init__.py", line 1, in <module>
    from litex.build.sim.platform import SimPlatform
  File "/home/wuhanstudio/litex/litex/build/sim/platform.py", line 14, in <module>
    from litex.build.sim import common, verilator
  File "/home/wuhanstudio/litex/litex/build/sim/common.py", line 17
    class SimDDROutput
                     ^
SyntaxError: invalid syntax
```

dayjaby, 2021-09-07:
Can you try again with a colon at the end of the line? If that works, feel free to create a PR :)

wuhanstudio, 2021-09-07:
Thanks, it's a small syntax error, and it works by adding a colon.

I just created a PR that fixes this small issue:

https://github.com/enjoy-digital/litex/pull/1028

enjoy-digital, 2021-09-07:
#1028 has been merged, thanks @wuhanstudio, @dayjaby.

closed 2021-09-07 by enjoy-digital