enjoy-digital/litex #1029: Interrupt support in Ibex CPU

navaneeth-cirel, 2021-09-08:
Hi,

Currently it looks like interrupt support is not yet added to Ibex CPU
1. Is there a plan on enabling this ?
2. How much amount of work is involved in this, maybe I can spend some time on this if I can get an outline of the steps involved

enjoy-digital, 2021-09-30:
Hi @navaneeth-cirel,

the Ibex CPU initial support does indeed not support interrupt. Adding interrupt support would be a nice contribution. To enable it, I would recommend studying some project with Ibex and interrupt enabled and then have a look at VexRiscv/Rocket integration in LiteX to understand how to connect the IRQs in the gateware and create the IRQ software support.

LiteX will automatically connect the interrupts in the SoC when the CPU has an `interrupt` signal: ex [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/core.py#L131) on Vex.

This signal then has to be connected to the CPU: ex [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/core.py#L144) on Vex (and eventually adapted if interrupts behave differently in Ibex).

The IRQ software support for Vex is defined in [irq.h](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/irq.h) which would have to be filled for Ibex [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/ibex/irq.h).

Happy to give you more directions/answer other questions if you want to have a look at this.

navaneeth-cirel, 2021-09-30:
Hi @enjoy-digital ,

Thank you for the pointers, I will take a look into this.

From an initial research I think looking at [CV32E40P](https://github.com/enjoy-digital/litex/blob/4563ccae18e28095baae1ea5c40844498b4b1d48/litex/soc/cores/cpu/cv32e40p/core.py#L437) would be easier because Ibex is more similar to it since both CV32E40P and Ibex initially came from the same project.

Dolu1990, 2021-09-30:
About VexRiscv, it implement the standard privilegied RISC-V spec for interrupts, but also add on the top the following implementation (which is used in the case of litex) : 
https://github.com/SpinalHDL/VexRiscv/blob/master/src/main/scala/vexriscv/plugin/ExternalInterruptArrayPlugin.scala#L6