enjoy-digital/litex #1036: CFU layout is not compatible with the generated VexRiscv core

GuzTech, 2021-09-13:
I have been using the CFU-Playground repository, and I want to create a simple test project in which I can easily create a CFU and hook it up to a LiteX SoC with a VexRiscv processor. However I have found that when generating the VexRiscv with a CFU, it does not have the `payload response ok` signal.

The CFU generated using the CFU-Playground repo does generate this signal, and LiteX [actually connects this signal](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/cpu/vexriscv/core.py#L293) to the generated VexRiscv, but since it does not exist it fails during synthesis. The CFU-Playground repo uses an older or custom version of LiteX, because this problem does not exist in their version.

For now, I've removed this signal from both LiteX and the CFU generator, and it works correctly then. Their project also has their own VexRiscv repository, but I don't know which one is newer or more correct. It seems their version is correct since upstream LiteX also expects this signal.

tcal-x, 2021-09-13:
Thanks for reporting @GuzTech , I'll take a look.

tcal-x, 2021-09-13:
> However I have found that when generating the VexRiscv with a CFU, it does not have the `payload response ok` signal.

@GuzTech are you rebuilding VexRiscv (using sbt and SpinalHDL), or using one of the pre-built ones?   
The prebuilt VexRiscv (CFU variants) should have the signal `CfuPlugin_bus_rsp_payload_response_ok`, for example https://github.com/tcal-x/pythondata-cpu-vexriscv/blob/cbb04d34fec970ebb660147932da0bb4bb5adce0/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v#L62

tcal-x, 2021-09-13:
@GuzTech , oh, I see what you mean, that signal is missing in the upstream repo.   That is a mistake.  
https://github.com/litex-hub/pythondata-cpu-vexriscv/blob/master/pythondata_cpu_vexriscv/verilog/VexRiscv_FullCfu.v

Thanks again for reporting this!

tcal-x, 2021-09-14:
@GuzTech -- another update.   The removal of `CfuPlugin_bus_rsp_payload_response_ok` was likely intentional -- there was some discussion about it, and it is not actually used in any designs currently.  It is my mistake that I didn't notice that the fork that CFU-Playground playground uses got out of sync with upstream.  Once the fate of that signal is decided, I should do whatever upstreaming is necessary so that I can stop using the fork and just use the upstream repo.

FYI @Dolu1990

GuzTech, 2021-09-21:
It was fixed with https://github.com/enjoy-digital/litex/pull/1039 so I am closing this issue. Thanks!

closed 2021-09-21 by GuzTech