enjoy-digital/litex #1042: soc/software/bios: The linker script breaks linkage with `--build-id` flag

ajelinski, 2021-09-23:
After running LiteDRAM's `test_examples.py` tests it turned out there's a linkage problem when GCC is configured with `--enable-linker-build-id` which makes it pass the `--build-id` flag to a linker. Using that option with current BIOS linker scripts ends up in such an error:
```
  <PATH>/ld.bfd.real: section .note.gnu.build-id LMA [0000000000000000,0000000000000023] overlaps section .text LMA [0000000000000000,0000000000005333]
  collect2: error: ld returned 1 exit status
  make: *** [<PATH>/litex/soc/software/bios/Makefile:60: bios.elf] Error 1
```

ajelinski, 2021-09-23:
CC @danc86

mithro, 2021-09-23:
@shenki - You were doing something here right?

mithro, 2021-09-23:
Might be related to https://github.com/enjoy-digital/litex/pull/1020 ?

ajelinski, 2021-09-23:
The testing was indeed conducted with an older LiteX revision (https://github.com/enjoy-digital/litex/commit/ab9f214a5ee4ab5bf96e19ff3520cf59092ad11a) so there should be no such error after that change.

Sorry for not testing the latest revision before creating this issue. I'll let you know after I check if this change is enough.

ajelinski, 2021-09-24:
As expected, the BIOS is buildable after making such a change in `common.mak`. Still, I'll leave closing the issue to @danc86 as he might want to add something to it.

danc86, 2021-09-27:
Thanks @ajelinski.

I don't think overriding `--build-id` to turn it off is the best approach, if the environment has explicitly requested it. Distros like Chrome OS generally want to inject a build id and expect that the resulting ELF executable will have the `.note.gnu.build-id` section, so that their debuginfo processing can do stuff with it.

More generally, non-loadable `.note` sections are used for injecting various other kinds of metadata into the executable, like Fedora's [annobin plugin](https://developers.redhat.com/blog/2018/02/20/annobin-storing-information-binaries#how_to_use_it). And DWARF debug sections fall into this category too, they are extra metadata kept in the ELF output even though they will not be included in the final firmware image.

The issue with `--build-id` seems to just be that the `.note.gnu.build-id` section is marked *loadable*, unlike the other kinds of metadata sections. As far as I can tell, this is done so that programs can extract their own build id at runtime. I even found [a blog post](https://interrupt.memfault.com/blog/gnu-build-id-for-firmware) about using it in embedded firmware to identify the version of code running, which is an interesting idea.

But since the Litex BIOS can't use the build id at runtime I would suggest to just mark it non-loadable in the linker script, so that it's kept in the ELF but doesn't cause an error by overlapping with .text and won't be copied to the final `bios.bin` image. I've sent PR #1044 for that.

Either way, the issue with `--build-id` doesn't happen with the latest `master` branch so we can definitely close this issue.

ajelinski, 2021-10-06:
I'm closing the issue since the problem is fixed and the change proposed by @danc86 has its own PR where it can be discussed.

closed 2021-10-06 by ajelinski