enjoy-digital/litex #1048: picolibc: Build broken on MacOS?

enjoy-digital, 2021-09-28:
Reported by @zyp: https://libera.irclog.whitequark.org/litex/2021-09-27#30903922;

```
if [ -d "/Users/zyp/Documents/orbtrace/deps/litex/litex/soc/software/libc/riscv" ]; then \
		cp /Users/zyp/Documents/orbtrace/deps/litex/litex/soc/software/libc/riscv/* /Users/zyp/Documents/orbtrace/deps/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/machine/riscv/ ;\
	fi
meson /Users/zyp/Documents/orbtrace/deps/pythondata-software-picolibc/pythondata_software_picolibc/data \
		-Dmultilib=false \
		-Dpicocrt=false \
		-Dthread-local-storage=false \
		-Dio-long-long=true \
		-Dformat-default=integer \
		-Dincludedir=picolibc/riscv64-unknown-elf/include \
		-Dlibdir=picolibc/riscv64-unknown-elf/lib \
		--cross-file cross.txt
WARNING: Unknown CPU family riscv, please report this at https://github.com/mesonbuild/meson/issues/new
The Meson build system
Version: 0.58.0
Source dir: /Users/zyp/Documents/orbtrace/deps/pythondata-software-picolibc/pythondata_software_picolibc/data
Build dir: /Users/zyp/Documents/orbtrace/build/orbtrace_mini_test/software/libc
Build type: cross build
Project name: picolibc
Project version: 1.7.2
C compiler for the host machine: riscv64-unknown-elf-gcc (gcc 10.2.0 "riscv64-unknown-elf-gcc (GCC) 10.2.0")
C linker for the host machine: riscv64-unknown-elf-gcc ld.bfd 2.35
C compiler for the build machine: cc (clang 10.0.1 "Apple LLVM version 10.0.1 (clang-1001.0.46.4)")
C linker for the build machine: cc ld64 450.3
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: riscv
Host machine cpu: vexriscv
Target machine cpu family: riscv
Target machine cpu: vexriscv
Compiler for C supports arguments -fno-stack-protector: YES 
Compiler for C supports arguments -fno-common: YES 
Compiler for C supports arguments -frounding-math: YES 
Program riscv64-unknown-elf-gcc-nm found: YES
Program scripts/duplicate-names found: YES (/Users/zyp/Documents/orbtrace/deps/pythondata-software-picolibc/pythondata_software_picolibc/data/scripts/duplicate-names)
Compiler for C supports link arguments -Wl,--defsym=_start=0: YES 
Compiler for C supports link arguments -Wl,-alias,main,testalias: NO 
Compiler for C supports function attribute alias: YES 
Compiler for C supports function attribute format: YES 
Configuring picolibc.specs using configuration

../../../../deps/pythondata-software-picolibc/pythondata_software_picolibc/data/meson.build:430:0: ERROR: "install_dir" must be specified when "install" in a configure_file is true

A full log can be found at /Users/zyp/Documents/orbtrace/build/orbtrace_mini_test/software/libc/meson-logs/meson-log.txt
make: *** [libc.a] Error 1
Traceback (most recent call last):
  File "/Users/zyp/Documents/orbtrace/./orbtrace_builder.py", line 103, in <module>
    main()
  File "/Users/zyp/Documents/orbtrace/./orbtrace_builder.py", line 92, in main
    builder.build(**trellis_argdict(args), run=args.build)
  File "/Users/zyp/Documents/orbtrace/deps/litex/litex/soc/integration/builder.py", line 302, in build
    self._generate_rom_software(compile_bios=use_bios)
  File "/Users/zyp/Documents/orbtrace/deps/litex/litex/soc/integration/builder.py", line 247, in _generate_rom_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/local/Cellar/python@3.9/3.9.5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/Users/zyp/Documents/orbtrace/build/orbtrace_mini_test/software/libc', '-f', '/Users/zyp/Documents/orbtrace/deps/litex/litex/soc/software/libc/Makefile']' returned non-zero exit status 2.

```

cc @kgugala, @michalsieron.

zyp, 2022-03-17:
Hi, I never tracked down what caused this issue, but it's not present anymore after updating to current pythondata-software-picolibc, so whatever it was it has been fixed in the meantime. This issue can be closed.

enjoy-digital, 2022-04-22:
Thanks @zyp for the feedback, let's close then.

closed 2022-04-22 by enjoy-digital