enjoy-digital/litex #1059: riscv64-unknown-gcc toolchain version 10.2.0-2020.12.8-x86_64-linux-centos error

aghaali-rs, 2021-10-06:
I am using RISCV64 GCC toolchain version 10.2.0-2020.12.8 for centos 7. I was initially at commit 694878a35a2f792ad195ff9c585d689467a76b1d which worked fine however I recently switched to commit cb6861e1c81e154b953d33b69d2fe1490195185d and now i get the error in the attached screenshot.

[image]
.
I am using a prebuild from https://github.com/sifive/freedom-tools/releases 
However, when using version 8.3.0-2019.08.0 for Linux it works fine on centos 7.

aghaali-rs, 2021-10-06:
A fresh install of LiteX and a fresh download of the prebuilt toolchain result in the error seen in the attached screenshot.
[image]

akioolin, 2021-10-06:
Hi, 

In my current system. the SiFive prebuild gnu cross gcc toolchain 10.1.0 could work very well in CFU-Playground which is enabled picolibc support. 

https://static.dev.sifive.com/dev-tools/freedom-tools/v2020.08/riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14.tar.gz

BR, Akio

aghaali-rs, 2021-10-12:
I tried the following four different versions of the toolchain:

- riscv64-unknown-elf-toolchain-10.2.0-2020.12.8-x86_64-linux-centos6
- riscv64-unknown-elf-toolchain-10.2.0-2020.12.8-x86_64-linux-ubuntu14
- riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-centos6
- riscv64-unknown-elf-gcc-10.1.0-2020.08.2-x86_64-linux-ubuntu14

Out of these four versions both 10.2.0 versions (centos6 and ubuntu14) had the same error however both 10.1.0 versions are working fine.

enjoy-digital, 2022-01-07:
The issue seems related to the 10.2 toolchain and a working toolchain should be used.

closed 2022-01-07 by enjoy-digital