enjoy-digital/litex #1062: Zephyr: riscv_litex drivers broken by 32-bit CSR Bus

mckaymatthew, 2021-10-12:
### The issue
It appears that when LiteX switched from 8-bit CSR bus to 32-bit CSR bus, the Zephyr drivers were not updated. 
The riscv_litex Zephyr port is not really operable in this situation, as the Zephyr kernel isn't getting timing ticks via the Timer module Event, as the Timer register map and configuration code is wrong. 
LiteX seems to [no longer support CSR buses less than 32 bits](https://github.com/enjoy-digital/litex/commit/4ee9c53f185ac6dd9c4aa69a7547e9bc037acc25), so the CSR bus cannot be set back to 8 bits (Removing this assert it builds, but the BIOS hangs)

### Driver status
The Zephyr drivers seem fall into two categories, those using the HAL [soc/riscv/litex/soc.h:litex_write](https://github.com/zephyrproject-rtos/zephyr/blob/main/soc/riscv/litex-vexriscv/soc.h), and those using the sys_write8. Those using the HAL may be easier to update, as they may only need register map updates

Drivers using the HAL:
- [gpio_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/gpio/gpio_litex.c)
- [i2c_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2c/i2c_litex.c)
- [i2s_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2s/i2s_litex.c) and [.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/i2s/i2s_litex.h)

Drivers needing partial-re-writes to use HAL:
- [hwinfo_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/hwinfo/hwinfo_litex.c)
- [eth_liteeth.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_liteeth.c)
- [litex_timer.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/timer/litex_timer.c)
- [pwm_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/pwm/pwm_litex.c)
- [uart_liteuart.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/serial/uart_liteuart.c) - Works, but may want updates to HAL anyway

Not sure:
- [lock_control_litex.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/clock_control/clock_control_litex.c)/[h](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/clock_control/clock_control_litex.h)

### Path forward
I wish to be friendly and respectful, I am entirely new to this ecosystem and would like to contribute in the form of fixing these drivers.

Before I do this I think there needs to be a discussion:
- Have I diagnosed this issue correctly?
- Do the original authors intend to fix this (AntMicro?), or is this is a good candidate for me to attempt to fix?
- What is the best way to get questions about LiteX platform answered (IRC can be slow, but works!). This started as a discussion on IRC, Florent asked me to create an issue. 
- Come up with test methodology for platforms I do not have (hwinfo, i2c, i2s)
- I initially created an [issue on the Zephyr repo](https://github.com/zephyrproject-rtos/zephyr/issues/39298) about the timer not working, before I understood the scope of the problem. Should I close this?

mithro, 2021-10-12:
PTAL -- @kgugala / @mateusz-holenko

mateusz-holenko, 2021-10-12:
Hi @mckaymatthew, thanks for reporting the issue.

At the time when the drivers were originally developed 8-bit CSR bus width was the default one and the code was developed to support that configuration.

As you correctly analyzed, since LiteX now uses 32-bit CSR bus width by default, drivers might not work out-of-the-box (although they are still usable when LiteX is configured with `--csr-data-width 8` [which as far as I know is still supported]).

We'll take a look at that and make sure that mainstream LiteX and Zephyr work fine together.

lachlansmith, 2022-04-03:
Hi has this issue been resolved? Simply trying to print "Can I keep Hello Worlding?" every 250 milliseconds and it is deadlocking, https://github.com/zephyrproject-rtos/zephyr/pull/39304

mckaymatthew, 2022-04-09:
@lachlansmith, it is still broken upstream. If you look at my (incomplete) pull request, the timer module is actually fairly simple to patch to work and should get your kernel cycling. The other thing you'll need to do is put your clock rate in proj.conf (In my case the sys clock rate is 55mhz,  CONFIG_SYS_CLOCK_HW_CYCLES_PER_SEC=55000000) for the timer to be accurate
I haven't needed any other drivers so I cannot comment, but it is reasonable to assume its all broken. 
More knowledgeable people expressed interest in working on this, so I abandoned my effort and just hacked up my kernel locally. I could still work this and I am still interested in contributing, @mateusz-holenko can you provide guidance?

lachlansmith, 2022-04-09:
Hi @mckaymatthew have you tried @mateusz-holenko suggestion to use ```--csr-data-width 8``` with any luck? Re-writing drivers isn't ideal, I'm about to give this a go.

michalsieron, 2022-04-11:
Hi @lachlansmith and @mckaymatthew, we are working on this. The solution is almost ready. There will be a PR opened soon.

lachlansmith, 2022-04-18:
Hi @michalsieron thanks for letting us know. I'm trying to use a Particle Argon with the Arty A7 to implement a fully-fledged Bluetooth stack over HCI SPI. I would use HCI UART but the driver doesn't support interrupts as yet. Could you confirm perhaps if the UART driver will support interrupts in the coming release?

michalsieron, 2022-04-28:
@lachlansmith, @mckaymatthew I opened two PRs in Zephyr repo (https://github.com/zephyrproject-rtos/zephyr/pull/45196 and https://github.com/zephyrproject-rtos/zephyr/pull/45198) and one in LiteX (https://github.com/enjoy-digital/litex/pull/1287), which solve this problem without sacrificing support for 8-bit CSRs.

enjoy-digital, 2022-05-27:
We can probably close this since #1287 has been merged. Thanks @mateusz-holenko, @michalsieron for looking at this.

closed 2022-05-27 by enjoy-digital