enjoy-digital/litex #1095: Unsupported RAM template

yetifrisstlama, 2021-11-02:
Using a Memory with 2 ports (both read and write) I get this error with Vivado v2021.1:
```
ERROR: [Synth 8-2914] Unsupported RAM template [.../gateware/test.v:4513]
ERROR: [Synth 8-5743] Unable to infer RAMs due to unsupported pattern.
```

# Litex code
```python
self.specials.mem = Memory(
    8 * D,
    mem_depth,
    name="capture_memory",
    init=[0x01, 0x02, 0x03, 0xDEADBEEF]
)
self.specials.p1 = p1 = self.mem.get_port(
    write_capable=True,
    # mode=READ_FIRST
)

self.sync += [
    # If(fsm.ongoing("DONE"),
    #     p1.adr.eq(0),
    If(p1.we,
        p1.adr.eq(p1.adr + 1),
    )
]

self.comb += [
    p1.dat_w.eq(sink.data),
    p1.we.eq(sink.valid),
    sink.ready.eq(1)
]
```

# Generated Verilog code:
```verilog
//------------------------------------------------------------------------------
// Memory capture_capture_memory: 512-words x 32-bit
//------------------------------------------------------------------------------
// Port 0 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
// Port 1 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
reg [31:0] capture_capture_memory[0:511];
initial begin
	$readmemh("capture_capture_memory.init", capture_capture_memory);
end
reg [8:0] capture_capture_memory_adr0;
reg [8:0] capture_capture_memory_adr1;
always @(posedge sys_clk) begin
	if (main_dp_capture_we)
		capture_capture_memory[main_dp_capture_adr] <= main_dp_capture_dat_w;
	capture_capture_memory_adr0 <= main_dp_capture_adr;
end
always @(posedge sys_clk) begin
	if (builder_sram0_we)
		capture_capture_memory[builder_sram0_adr] <= builder_sram0_dat_w;
	capture_capture_memory_adr1 <= builder_sram0_adr;
end
assign main_dp_capture_dat_r = capture_capture_memory[capture_capture_memory_adr0];
assign builder_sram0_dat_r = capture_capture_memory[capture_capture_memory_adr1];
```

I tried the port in `READ_FIRST` and `WRITE_FIRST` mode with the same result.
Possibly a regression, as Memory was inferred fine a few weeks ago.

enjoy-digital, 2021-11-02:
Thanks @yetifrisstlama for the feedback, I'll look at it.

enjoy-digital, 2021-11-02:
@yetifrisstlama: When testing with upstream LiteX and 2021.08, the only difference I can see (except the rename of the address registers that has no functional impact) is the location of the memory initialization:
[image]

Could you do a test with LiteX 2021.08 or try to move the memory initialization (https://github.com/enjoy-digital/litex/blob/master/litex/gen/fhdl/memory.py#L80-L82) to see if it's really a regression and if so if it's related to the move of the memory initialization? Thanks.

yetifrisstlama, 2021-11-03:
Many thanks @enjoy-digital 

I think I was wrong about the regression. Also I take it back that both modes lead to equal results. In fact, things are much stranger ...

My Vivado recognizes these two as a BRAM template successfully:
```verilog
//------------------------------------------------------------------------------
// Memory capture_capture_memory: 512-words x 32-bit
//------------------------------------------------------------------------------
// Port 0 | Read: Sync  | Write: Sync | Mode: Read-First  | Write-Granularity: 32 
// Port 1 | Read: Sync  | Write: Sync | Mode: Read-First  | Write-Granularity: 32 
reg [31:0] capture_capture_memory[0:511];
reg [31:0] capture_capture_memory_dat0;
reg [31:0] capture_capture_memory_dat1;
always @(posedge sys_clk) begin
	if (main_dp_capture_we)
		capture_capture_memory[main_dp_capture_adr] <= main_dp_capture_dat_w;
	capture_capture_memory_dat0 <= capture_capture_memory[main_dp_capture_adr];
end
always @(posedge sys_clk) begin
	if (builder_sram0_we)
		capture_capture_memory[builder_sram0_adr] <= builder_sram0_dat_w;
	capture_capture_memory_dat1 <= capture_capture_memory[builder_sram0_adr];
end
assign main_dp_capture_dat_r = capture_capture_memory_dat0;
assign builder_sram0_dat_r = capture_capture_memory_dat1;
```

```verilog
//------------------------------------------------------------------------------
// Memory capture_capture_memory: 512-words x 32-bit
//------------------------------------------------------------------------------
// Port 0 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
// Port 1 | Read: Sync  | Write: Sync | Mode: Read-First  | Write-Granularity: 32 
reg [31:0] capture_capture_memory[0:511];
reg [8:0] capture_capture_memory_adr0;
reg [31:0] capture_capture_memory_dat1;
always @(posedge sys_clk) begin
	if (main_dp_capture_we)
		capture_capture_memory[main_dp_capture_adr] <= main_dp_capture_dat_w;
	capture_capture_memory_adr0 <= main_dp_capture_adr;
end
always @(posedge sys_clk) begin
	if (builder_sram0_we)
		capture_capture_memory[builder_sram0_adr] <= builder_sram0_dat_w;
	capture_capture_memory_dat1 <= capture_capture_memory[builder_sram0_adr];
end
assign main_dp_capture_dat_r = capture_capture_memory[capture_capture_memory_adr0];
assign builder_sram0_dat_r = capture_capture_memory_dat1;
```

But I get the aforementioned error for these two:
```verilog
//------------------------------------------------------------------------------
// Memory capture_capture_memory: 512-words x 32-bit
//------------------------------------------------------------------------------
// Port 0 | Read: Sync  | Write: Sync | Mode: Read-First  | Write-Granularity: 32 
// Port 1 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
reg [31:0] capture_capture_memory[0:511];
reg [31:0] capture_capture_memory_dat0;
reg [8:0] capture_capture_memory_adr1;
always @(posedge sys_clk) begin
	if (main_dp_capture_we)
		capture_capture_memory[main_dp_capture_adr] <= main_dp_capture_dat_w;
	capture_capture_memory_dat0 <= capture_capture_memory[main_dp_capture_adr];
end
always @(posedge sys_clk) begin
	if (builder_sram0_we)
		capture_capture_memory[builder_sram0_adr] <= builder_sram0_dat_w;
	capture_capture_memory_adr1 <= builder_sram0_adr;
end
assign main_dp_capture_dat_r = capture_capture_memory_dat0;
assign builder_sram0_dat_r = capture_capture_memory[capture_capture_memory_adr1];
```

```verilog
//------------------------------------------------------------------------------
// Memory capture_capture_memory: 512-words x 32-bit
//------------------------------------------------------------------------------
// Port 0 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
// Port 1 | Read: Sync  | Write: Sync | Mode: Write-First | Write-Granularity: 32 
reg [31:0] capture_capture_memory[0:511];
reg [8:0] capture_capture_memory_adr0;
reg [8:0] capture_capture_memory_adr1;
always @(posedge sys_clk) begin
	if (main_dp_capture_we)
		capture_capture_memory[main_dp_capture_adr] <= main_dp_capture_dat_w;
	capture_capture_memory_adr0 <= main_dp_capture_adr;
end
always @(posedge sys_clk) begin
	if (builder_sram0_we)
		capture_capture_memory[builder_sram0_adr] <= builder_sram0_dat_w;
	capture_capture_memory_adr1 <= builder_sram0_adr;
end
assign main_dp_capture_dat_r = capture_capture_memory[capture_capture_memory_adr0];
assign builder_sram0_dat_r = capture_capture_memory[capture_capture_memory_adr1];
```

This is on current litex master c13be522cef45f021de87d17d574e975920df535 and I get identical results on 2021.08.

yetifrisstlama, 2021-11-06:
This code does reproduce the error for me:

```python
# from migen import *
from litex.soc.integration.builder import Builder
from litex.soc.interconnect.wishbone import SRAM
from litex_boards.targets.berkeleylab_marble import BaseSoC
from migen import *

class TestSoc(BaseSoC):
    def __init__(self):
        BaseSoC.__init__(
            self,
            cpu_type=None,
            integrated_main_ram_size=8,
            integrated_sram_size=0,
            uart_name='uartbone',
            uart_baudrate=115200
        )

        # These two cases get recognized as dual port BRAM in Vivado v2021.1
        # mode_p1, mode_p2 = READ_FIRST, READ_FIRST
        # mode_p1, mode_p2 = WRITE_FIRST, READ_FIRST

        # These two cases fail to be recognized with the following error
        # ERROR: [Synth 8-2914] Unsupported RAM template [berkeleylab_marble.v:1383]
        # ERROR: [Synth 8-5743] Unable to infer RAMs due to unsupported pattern.

        # mode_p1, mode_p2 = READ_FIRST, WRITE_FIRST
        mode_p1, mode_p2 = WRITE_FIRST, WRITE_FIRST

        self.specials.test_mem = Memory(
            32,
            512,
            init=[0x01, 0x02, 0x03, 0xDEADBEEF]
        )

        self.specials.p1 = p1 = self.test_mem.get_port(
            write_capable=True,
            mode=mode_p1,
            # we_granularity=8  # doesn't make a difference
        )

        # Write some dummy data, else Vivado will optimize the port away
        counter = Signal(24)
        self.sync += [
            p1.we.eq(0),
            counter.eq(counter + 1),
            If(counter == 0,
                p1.we.eq(1),
                p1.adr.eq(p1.adr + 1),
            )
        ]

        self.comb += [
            p1.dat_w.eq(0xdeadbeef)
        ]

        # Need to hack the mode argument into the SRAM class
        self.submodules.sram = SRAM(self.test_mem, mode=mode_p2)
        self.register_mem("test_ram", 0x10000000, self.sram.bus, 512 * 32)

def main():
    soc = TestSoc()
    builder = Builder(
        soc,
        compile_gateware=True,
        csr_csv='./build/csr.csv'
    )
    builder.build()

if __name__ == "__main__":
    main()
```

I've done some experimentation with manually changing the generated Verilog code to look more like the Vivado language template (Write First Mode w/ Byte-write). But I haven't had much success yet. In fact, the only way I could get Vivado to recognize the BRAM with 2 ports in `WRITE_FIRST` mode was by copy-pasting and instantiating the whole template:

```verilog
//  Xilinx True Dual Port RAM Byte Write Write-First Dual Clock RAM
  xilinx_true_dual_port_write_first_byte_write_2_clock_ram #(
```

This is really quite frustrating to debug, Vivado is very picky about the BRAM template but on the other hand doesn't provide any helpful error message at all how to match it.

enjoy-digital, 2021-11-06:
Thanks @yetifrisstlama for the minimal repro. Now that the memory generation has been reworked, it will be easier to customize memory pattern generation. I'll also have a look next week.