enjoy-digital/litex #1097: Make Firmware error in lab004 from fpga_101

hyde796, 2021-11-07:
Hello, does anyone have problems compiling the firmware from the lab004?
I'm using KDE Neon based on Ubuntu 18 and the last Litex and RiscV toolchain version, my target is an arty_s7.

When I try to compile de firmware I get this error

Makefile:18: update target 'firmware.elf' due to: isr.o main.o
echo " LD " firmware.elf && riscv64-unknown-elf-ld -nostdlib -nodefaultlibs -Wl,--no-dynamic-linker -Wl,--build-id=none -MD -MP -Os -march=rv32im -mabi=ilp32 -D__vexriscv__ -g3 -fomit-frame-pointer -Wall -fno-builtin -fno-stack-protector -flto -I/home/hyde/Digital/Litex/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/tinystdio -I/home/hyde/Digital/Litex/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include -I/home/hyde/Digital/Litex/litex/litex/soc/software/libbase -I/home/hyde/Digital/Litex/litex/litex/soc/software/include -I/home/hyde/Digital/Litex/litex/litex/soc/software -I/home/hyde/Digital/FPGA/fpga_101/lab004_s7/build/software/include -I/home/hyde/Digital/FPGA/fpga_101/lab004_s7/build/software/include/../libc -I/home/hyde/Digital/Litex/litex/litex/soc/cores/cpu/vexriscv -fexceptions -Wstrict-prototypes -Wold-style-definition -Wmissing-prototypes -L/home/hyde/Digital/FPGA/fpga_101/lab004_s7/build/software/include
-T linker.ld
-N -o firmware.elf
../build//software/libbase/crt0.o
isr.o main.o
-L../build//software/libbase
-L../build//software/libcompiler_rt
-lbase-nofloat -lcompiler_rt
LD firmware.elf
riscv64-unknown-elf-ld: unrecognised emulation mode: abi=ilp32
Supported emulations: elf64lriscv elf32lriscv
make: *** [Makefile:18: firmware.elf] Error 1

So far I can't fix this issue

hyde796, 2021-11-09:
Hi @enjoy-digital,

Yes, I am using the toolchain provided by Litex in the install script, I also use the nexys4ddr as platform and  target a vexriscv and a picorv32 cpu but in both cases I have the same result.

I also try with an lm32 core, in this case I have to correct the **"/pythondata-software-picolibc/pythondata_software_picolibc/data/newlib/libc/include/machine/endian.h"** file in  because it has some extra ")" ,  but also fails at using the LD to generate the **firmware.elf** file in this case with "lm32-elf-ld"

enjoy-digital, 2022-03-29:
This seems to be a toolchain installation issue, this would need to be investigated on the developer's machine.

closed 2022-03-29 by enjoy-digital