enjoy-digital/litex #1099: How to add extra UARTs to core

openpowerwtf, 2021-11-10:
I'm trying to add a second UART accessible by the core.  I tried a few ways but can't get both CSR and logic to be generated.  What's the proper way to do this?

```
# 1
self.submodules.uart_1 = UARTWishboneBridge(pins, sys_clk_freq, baudrate=115200)
self.add_wb_master(self.uart_1.wishbone)
# 2
self.submodules.uart_1 = UART(UARTPHY(pins, sys_clk_freq, 115200))
# 3
self.submodules.uart_1 = UARTBone(UARTPHY(pins, sys_clk_freq, 115200), sys_clk_freq) 

self.add_csr('uart_1')
```

mithro, 2021-11-10:
You haven't provided enough information to really know what is going wrong here. Can you link to your full source code and what you are seeing happening?

The things below are just random guesses and I have pretty low confidence in them.

My current guess with example 2 and 3 is that you want to add the `UARTPHY` as a submodule and then hand it to the `UART` / `UARTBone` modules like shown below;
https://github.com/enjoy-digital/litex/blob/ddcac462c3420f01a4f415c1d6bd7e1001b8227d/litex/soc/integration/soc.py#L1238-L1240

I'm not sure what is going on with example 1 -- one guess would be that you do not have any CPU / bridge in your SoC, so the CSR bus just doesn't exist. If you just want a bridge with no-CPU then you need to do something like
```python
        kwargs['cpu_type'] = None
        BaseSoC.__init__(self, platform, *args, with_uart=False, **kwargs)

        self.add_cpu_or_bridge(UARTWishboneBridge(platform.request("serial"), self.clk_freq, baudrate=19200))
        self.add_wb_master(self.cpu_or_bridge.wishbone)
```

openpowerwtf, 2021-11-11:
It seems like a CSR region gets allocated but it doesn't show up at the end (I think example 1 was before I realized the bridge was for an external host).  I put some info here:

https://git.openpower.foundation/cores/a2p/src/branch/master/build/litex/litex-1099

mithro, 2021-11-11:
@openpowerwtf - You are doing a lot of very weird things in your `a2p_cmod7_uarts.py` file @ https://git.openpower.foundation/cores/a2p/src/branch/master/build/litex/litex-1099/master/a2p_cmod7_uarts.py#L112 -- Are you on the `#litex` IRC channel, it is probably easier to help you via chat.

openpowerwtf, 2021-11-11:
Yeah, it's lonely being the last one in the universe doing ppc32be on Litex.   It does appear they let me on the #litex channel.

openpowerwtf, 2021-11-12:
@mithro - thanks!

Explicitly adding the  PHY CSR works - did a simple test with lxterm on both UARTs. 

```
self.submodules.uart_1_phy = RS232PHY(pins, sys_clk_freq, with_dynamic_baudrate=True)
self.add_csr('uart_1_phy')
self.submodules.uart_1 = UART(phy=self.uart_1_phy)
self.add_csr('uart_1')

# or
       
self.submodules.uart_1_phy = UARTPHY(pins, sys_clk_freq, 115200)
self.add_csr('uart_1_phy')
self.submodules.uart_1 = UART(phy=self.uart_1_phy)
self.add_csr('uart_1')
```

closed 2021-11-12 by mithro

russo00, 2025-04-15:
I added a second UART, and it transmits data to the SIM800 device correctly. However, reception is not working properly — incoming data on UART2 is not recognized. I have the lines #self.add_csr('uart_2_phy') and #self.add_csr("uart2") commented out. If I uncomment them, the main primary UART stops working, and I can no longer access the LiteX console. What could be the issue?

# Second UART for SIM800
from litex.soc.cores.uart import UART, UARTPHY
serial2_pads = platform.request("serial2")

self.submodules.uart2_phy = UARTPHY(pads=serial2_pads, clk_freq=sys_clk_freq, baudrate=115200)
#self.add_csr('uart_2_phy')
self.submodules.uart2 = UART(self.uart2_phy)
#self.add_csr("uart2")