enjoy-digital/litex #1101: Separate BIOS generation and bitstream upload

developandplay, 2021-11-12:
Currently regardless of whether --build or --load is executed we generate the BIOS.
However now that I'm trying to use my Raspberry Pi for CI automation this becomes an issue.
As far as I can tell on ARM the RISC-V Toolchain has no official prebuilt binaries.
Building the toolchain on the Raspberry Pi is not possible due to resource constraints.
Therefore it would probably be easiest to separate the BIOS generation such that it is independent from the --load command.
Then on a more powerful machine we can build bitstream, buildroot and BIOS while Raspberry Pi just uploads everything to the FPGA.
Alternatively of course we could setup some buildjob for the RISC-V Toolchain on Arm or convince sifive to do so.

developandplay, 2021-11-12:
Right now I'm using openocd to just upload the bitstream on the RPi. While that works perfectly fine I would expect --load to not rebuild the BIOS by default.

enjoy-digital, 2021-11-12:
Thanks @developandplay for the feedback. I'm going to spend some time improving the command line parameters soon and will try to address this. For now you can also use `--no-compile-software` to avoid the BIOS compilation.

kaolpr, 2021-12-07:
Actually, it's not only BIOS compilation. E.g. when you only want to load firmware Verilog output is also overwritten (and for most targets its unavoidable) and as build-time parameters may differ from load-time parameters, output file may be different. It's not critical, however it may load to some confusion when analyzing Verilog output after one loaded it to the hardware.

enjoy-digital, 2022-05-27:
Thanks @developandplay and @kaolpr for the feedback. The behavior has been changed recently and nothing is no longer generated when `--build` is not passed. (it just now only check the design syntax and display SoC information but does not generate software or gateware files).

closed 2022-05-27 by enjoy-digital