enjoy-digital/litex #1105: litex_term: Allow 256/128 bytes payload in upload_calibration

enjoy-digital, 2021-11-19:
The upload_calibration has been limited to 64 bytes payload for now: https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_term.py#L375

Supporting 256/128 is useful to speed-up upload over LitePCIe with liteuart driver (180KB/s @ 64-bytes to ~600KB/s @ 256-bytes), but this first has to be tested with the other UARTs.

cklarhorst, 2021-11-23:
@enjoy-digital 
Hi, I just run a quick test. I'm using a FTDI_FT231X_USB_UART device with a bitrate of 1000000.
My repo is on the master branch (aecfbc06098).
I get (without any errors):
inter-frame: 10.00us, length:  64  --> 86.0 KB/s
inter-frame: 10.00us, length: 128 --> 91.6 KB/s
inter-frame: 10.00us, length: 251 --> 94.6 KB/s

enjoy-digital, 2021-11-23:
Thanks @cklarhorst for the feedback, this is useful to know it's working fine with this configuration. I also want to validate specific configurations that were behaving differently:
- Over USB ACM (as used on OrangeCrab for example).
- Over USB FIFO (as used on NexysVideo for example).

I'll try to do this soon, we'll then be able to add 256/128 bytes support to litex_term.