enjoy-digital/litex #1106: litex: Add Fedora installation instructions.

enjoy-digital, 2021-11-19:
The installation guide on Linux only provide instructions for Ubuntu and needs to be adapted a little bit on Fedora machines.

Here are just some notes:
- glibc-devel.i686 needs to be installed.
- libevent-dev -> libevent-devel.
- libjson-c-dev ->json-c-devel.
- `./litex_setup.py --gcc=riscv` is also installing the `ubuntu` toolchain and should use the `centos` one.

gsomlo, 2021-11-19:
Why do you think the `i686` version of `glibc-devel` is needed? Seems to work fine on my end with the native `x86_64` version. Something about the precompiled riscv toolchain maybe? (in which case we can build and publish a prebuilt x86_64 toolchain, e.g. http://www.contrib.andrew.cmu.edu/~somlo/BTCP/RISCV-20210921gitb39e361.tar.xz, built from upstream https://github.com/riscv/riscv-gnu-toolchain)

BTW, the [linux-on-litex-rocket](https://github.com/litex-hub/linux-on-litex-rocket) instructions are fedora-based (see the prerequisites section)...

enjoy-digital, 2021-11-19:
Thanks @gsomlo, that's useful to know Linux-on-LiteX-Rocket's build is Fedora-based. I was just taking some notes here from an installation I did on a Fedora machine to avoid forgetting this.... I'll have a closer look for ` glibc-devel.i686`.

gsomlo, 2021-11-19:
On Fri, Nov 19, 2021 at 07:07:49AM -0800, enjoy-digital wrote:
> Thanks @gsomlo, that's useful to know Linux-on-LiteX-Rocket's build is
> Fedora-based.
 
Also, I've been using Fedora for many years now as my primary OS, and
I'm currently acting as package maintainer for several things in their
official repo that may be relevant to LiteX: yosys, trellis, icestorm,
nextpnr, and migen -- and keep them fairly up to date w.r.t. their
upstream github sources. Not sure nmigen is ready to be a distro
package, but if so, and if LiteX has any near-term plans on migrating
to it, I can probably make it an official Fedora package as well.

I can (at least) help you test whatever fedora-specific steps you plan
to recommend in the LiteX documentation, just LMK how I can help :)

enjoy-digital, 2021-11-26:
Thanks @gsomlo, sorry for the late answer. I just redid the installation on a Fedora machine:` ./litex_setup.py --gcc=riscv` will now use the centos6 version on a Fedora machine with: https://github.com/enjoy-digital/litex/commit/0503188661d7f6c04fba99b478ed2447cbe5db3c and it seems the only things  that would simplify the installation would be an update of verilator in the official repo since seems too old: I have to re-compile it, are you also the package maintainer for verilator?

Just logging the installation steps to update the doc later:
```
sudo dnf install python3-pip
wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
chmod +x litex_setup.py
sudo ./litex_setup.py --install
./litex_setup.py --gcc=riscv
sudo pip3 install meson ninja

sudo dnf install autoconf libevent-devel json-c-devel zlib-devel flex bison gcc-c++
git clone https://github.com/verilator/verilator
cd verilator
autoconf
./configure
make -j$(nproc)
sudo make install
```

gsomlo, 2021-11-26:
On Fri, Nov 26, 2021 at 06:10:22AM -0800, enjoy-digital wrote:
> Thanks @gsomlo, sorry for the late answer. I just redid the installation on a
> Fedora machine: ./litex_setup.py --gcc=riscv will now use the centos6 version
> on a Fedora machine with: 0503188 and it seems the only things that would
> simplify the installation would be an update of verilator in the official repo
> since seems too old: I have to re-compile it, are you also the package
> maintainer for verilator?

Sadly I am not... :) Verilator in Fedora is currently at 4.108 (from
January 2021, apparently). I notice they're up to 4.215 as of right
now, on github.

Do you have any specific reason to need a newer version? With that
information, we could email the maintainer and suggest/request an
update (this could be done officially via Bugzilla, or just an
informal email inquiry, depending on the severity of the issue...
 
> Just logging the installation steps to update the doc later:
> 
> sudo dnf install python3-pip
> wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
> chmod +x litex_setup.py
> sudo ./litex_setup.py --install
> ./litex_setup.py --gcc=riscv

> sudo pip3 install meson ninja

This should rather be `dnf install meson`! It will pull in
`ninja-build` as a dependency, and that's worked OK for me
in building the picolibc-based bios.

There's a whole "ethos" in Fedora of avoiding "side-loaded"
(via e.g., pip) packages in favor of distro-packaged ones
whenever possible. It's sometimes a bit annoying when the
priority is fast prototyping, but I also value the long-term
benefits of keeping things under some semblance of order and
control... :)
 
> sudo dnf install autoconf libevent-devel json-c-devel zlib-devel flex bison gcc-c++
> git clone https://github.com/verilator/verilator
> cd verilator
> autoconf
> ./configure
> make -j$(nproc)
> sudo make install

enjoy-digital, 2021-11-26:
For Verilator, I was hitting https://www.reddit.com/r/FPGA/comments/op24te/when_gccc_with_stdgnu14_and_a_lot_of_warning/ and so there that the issue was fixed with 4.110 so just clone it and recompiled it, so it could indeed be worth updating the official version. 

For meson, we are using pip3 version with Ubuntu since the official version is too old and I just tried to keep things similar when possible with Fedora. I'll also test with dnf install meson, thanks.

gsomlo, 2021-11-26:
On Fri, Nov 26, 2021 at 07:15:10AM -0800, enjoy-digital wrote:
> For Verilator, I was hitting https://www.reddit.com/r/FPGA/comments/op24te/
> when_gccc_with_stdgnu14_and_a_lot_of_warning/ and so there that the issue was
> fixed with 4.110 so just clone it and recompiled it, so it could indeed be
> worth updating the official version.

https://bugzilla.redhat.com/show_bug.cgi?id=2026957

enjoy-digital, 2021-11-26:
Thanks @gsomlo!